ascillato/Tasmota_KNX #25: Updating existing Tasmota firmware to the MOD

johannesbonn, 2018-04-02:
Hello,
I have found your repository and it sounds awesome! Great job.
My question is: Is it possible to change to your Tasmota Mod by using the existing ota Update routine with the web interface of an original Tasmota firmware? 
This would be a very eas way to change to your Mod.

Have a nice weekend

Johannes

ascillato, 2018-04-02:
Hi,

Yes can be updated using OTA. This Firmware is exactly the same Tasmota. It is in sync with the original tasmota.
The only diference is the addition of the KNX driver.

This repo is for developing the driver that connects TASMOTA to the KNX Library (ESP_KNX_IP)

Now it is in development. I'm working on it. It is not finished at this moment.

As soon as I have a working version I will update this repo and also make a Pull Request to the Original Tasmota to add this KNX Driver.

ascillato, 2018-04-02:
At this moment the communication works but it loses the configuration when the device is resetted (because that part is not implemented yet - see readme)

I'm very excited with the preliminary tests because it works really good and it is very nice to see several sonoff communicating each other directly, like using the button of one device to turn on other devices without the need of a mqtt server or something else.

johannesbonn, 2018-04-02:
Perfect, I am waiting for you! Thanks for your work!

closed 2018-04-03 by johannesbonn