ascillato/Tasmota_KNX #44: Ota Upgrade failed

johannesbonn, 2018-04-10:
Hi,
I have tested a ota upgrade with one of my s20 devices (version 5.12.0) but unfortunaly it failed. There is no error or something like that. The browser loses the connection. After replug the device every thing runs but with the old installed version. Only the upgrade to the minimal setup runs, but the wlan connection is so unstable, that I can not upgrade again. I don‘t know if it  has something to do with my setup???

I am very excited to test your upgrade. I wrote again after some more tests.

Thank you for your work!

Johannes

ascillato, 2018-04-11:
If still does not work, you can try reset configuration (from the menu or pressing for 40 secs the button) and then upgrade the firmware.

johannesbonn, 2018-04-13:
Hello,

sorry for my late reply. Unfortunately I have no luck. After every upgrade the devices (I have tested it with 3 different S20 plugs) go offline and after some minutes I can reach the devices but with the old firmeware. All devices have firmware Version 5.12.0 installed. No difference between the normal (not 4M) and minimal firmware.
What is the status to integrate your knx modification to the offical tasmota release?

Do you have any hint for me.

ascillato, 2018-04-15:
Hi,

Sorry for the late response.

I had no problems updating using the provided firmware. But for 1 device with an old firmware I had to reset the configuration before upgrading the firmware. In the lasts versions, the memory map of the tasmota firmware has changed, so, sometimes the old configuration just don't work. Most of the time the OTA works for me, but with old firmwares sometimes I have to reset config.

1- Did you try the reset config before upgrade?
2- If you do the OTA with the v5.11.0 from Tasmota. It works?

Anyway, I'm going to release the v5.12.0i that is the same Tasmota but with the KNX Driver. We can also try with that.

Theo say that he is going to integrate the Pull Request I made. So, may be in sometime we will see that.

johannesbonn, 2018-04-15:
So on, new run, but without any luck. After reseting the config I get te following error: Upload Failed

Upload buffer miscompare.

What does it mean?

Then my second try: upgrade to minimal Firmware and then upgrade to normal one......
same error message:
Upload Failed

Upload buffer miscompare

Any ideas?

ascillato, 2018-04-15:
Ok, we have now an error message to work on. That's good. Let me see that. I writing to you soon. Thanks

ascillato, 2018-04-17:
Hi,

@johannesbonn , can you test the firmware from the [new release v5.12.0l](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.12.0l)?

johannesbonn, 2018-04-18:
Hello,
sorry, but I get the same error message: "Upload Failed/Upload buffer miscompare"
I have tested two different Webbrowsers (edge/Firefox).
Any idea?

Thank you for your effort!

ascillato, 2018-04-19:
Hi,

Have you tried the [v5.12.0](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v5.12.0) from Tasmota ?

If that do not work, you will need to upload the firmware by wire, not OTA.

If that works, Try the [sonoff_KNX-EN-xxs.bin](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/download/v5.12.0l/sonoff_KNX-EN-xxs.bin) that is a Tasmota with KNX but with few sensors.

ascillato, 2018-04-19:
NEWS:

Theo has integrated this KNX modification into Tasmota!

At this moment this repository is the same as the Original Tasmota.

ascillato, 2018-04-19:
Hi,

I compile the new Tasmota Version that includes KNX [v5.12.0m](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.12.0m)

We can try this:

1 - Please try to upload by OTA the old [Tasmota v5.12.0](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v5.12.0) ?

_( If that do not work, you will need to upload the firmware by wire, not OTA )_

2 - If that works, please try the [sonoff_KNX-EN-xxs.bin](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/download/v5.12.0m/sonoff_KNX-EN-xxs.bin) that is a Tasmota with KNX but with few sensors.

3 - And finally try the [sonoff_KNX-EN.bin](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/download/v5.12.0m/sonoff_KNX-EN.bin)

johannesbonn, 2018-04-19:
Step 3. makes no problems. An update with the existing version 5.12.0 was ok.
Today evening I will give it a try with the few sensor version.

ascillato, 2018-04-19:
Due to the ESP8266 Board Library v2.4.1 and v2.4.0 have issues regarding sleep, I have downgraded KNX to use v2.3.0 as Theo required.

That's why the new release: [v5.12.0n](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.12.0n)

Please, test this new release on your devices.

Thanks

johannesbonn, 2018-04-19:
Hi ascillato,

sorry, but both versions, the new one from today (index n) and the Version with few sensors failed,
everytime the same error message: Upload Failed/Upload buffer miscompare.

I don't understand the Problem? Is there a Probleme with buffer size?

Regards Johannes

ascillato, 2018-04-19:
Hi,

I test the OTA on several devices. In just one that had an old Tasmota Version gave me the same error. In that device I could not upload by OTA neither the same old version. I don't know why. In that device I finally had to download the firmware using the FTDI cable as it was a new Itead Sonoff.

How do you download the firmware the first time to your devices? by FTDI cable?

ascillato, 2018-04-19:
Another thing we can try is to download to the device:

1) The [**sonoff-minimal**](https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.12.0/sonoff-minimal.bin) firmware from tasmota [v5.12.0](https://github.com/arendst/Sonoff-Tasmota/releases/tag/v5.12.0). 
2) Then, the [**sonoff-minimal**](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/download/v5.12.0n/sonoff_minimal.bin) firmware from today release [v5.12.0n](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.12.0n)
3) Finally, the **Sonoff_KNX** firmware in the language you prefer from today release [v5.12.0n](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.12.0n)

johannesbonn, 2018-04-19:
Hi?
I upload the bin file with the webbrowser. With the official Tasmota files there is no problem.

ascillato, 2018-04-19:
good, and with the [sonoff-minimal](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/download/v5.12.0n/sonoff_minimal.bin) from today release?

ascillato, 2018-04-19:
I mean downloading sonoff-minimal from today using the sonoff-minimal from tasmota on your device

ascillato, 2018-04-21:
Hi,

New Idea to Test.

Please, try any of the compiled firmwares from [reloxx13/Sonoff-Tasmota-Modified](https://github.com/reloxx13/Sonoff-Tasmota-Modified/releases).

These ones are the new Tasmota version but without KNX.

If work, please after that, try the KNX ones of my firmwares.

ascillato, 2018-04-21:
Hi @johannesbonn ,

I think I found the problem !!!!  The new Original Tasmota is bigger than was on v5.12.0.

So, the firmwares I was releasing were bigger than 510KB. So, as explained on [Tasmota Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade), OTA will not work on 1MB devices. The firmware upgrade will work if you upload by cable or if you have a really small minimal firmware for doing the OTA.

So, I compile from the default Tasmota with KNX but without:
 - I2C sensors
 - Sunrise calculation
 - Domoticz support
 - mDNS Discorvery
 - HomeAssistant Discovery
 - Software Serial Bridge
 - IR TX and RX

So, for matching to the released versions with Tasmota, I have deleted the previous releases I made, and I release the [v5.12.0m](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.12.0m)

Please, try this new release.

ascillato, 2018-04-22:
Hi,

I tested on a Sonoff Bridge and OTA works with the new Sonoff-Tasmota_KNX release !!!!

Please confirm that also work for you

Thanks for the patience

johannesbonn, 2018-04-22:
confirmed, perfect!!!!! Thank you!

johannesbonn, 2018-04-22:
what was the Problem?

ascillato, 2018-04-22:
Great!

The problem was the size of the firmware.

closed 2018-04-22 by ascillato