ascillato/Tasmota_KNX #82: No response to KNX telegramms after about 15 minutes. Status telegramms to KNX are still working

RocketSience, 2018-05-01:
I am using an Enertex KNX router.
It does not matter from witch source the telegramms to the sonoff are comming, i used also an USB-KNX Interface.

After resetting the sonoff, its working (receiving) again for about 15minutes, after that only one direction ist working (sonoff -> knx).

- [x] What hardware you are using
Sonoff Basic, Enertex KNX router, EDOMI
- [x] Provide the output of command ``status 0``
`
18:25:37 CMD: status 0
18:25:37 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:25:37 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:53:51","StartupUTC":"2018-05-01T16:31:46","Sleep":0,"BootCount":6,"SaveCount":97,"SaveAddress":"FB000"}}
18:25:37 RSL: STATUS2 = {"StatusFWR":{"Version":"5.13.0","BuildDateTime":"2018.05.01 06:01:14","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:25:37 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxx",""],"TelePeriod":300,"SetOption":["02008001","55818000"]}}
18:25:37 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:25:37 RSL: STATUS5 = {"StatusNET":{"Hostname":"SONOFF_001","IPAddress":"192.168.0.33","Gateway":"192.168.0.253","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.253","Mac":"EC:FA:BC:10:56:38","Webserver":2,"WifiConfig":3}}
18:25:37 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue May 01 17:25:37 2018","Local":"Tue May 01 18:25:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:29","Sunset":"20:05"}}
18:25:37 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-05-01T18:25:37"}}
18:25:37 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-05-01T18:25:37","Uptime":"0T00:53:51","Vcc":3.161,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","RSSI":68,"APMac":"9C:C7:A6:49:BB:31"}}}
18:27:55 RSL: STATE = {"Time":"2018-05-01T18:27:55","Uptime":"0T00:56:09","Vcc":3.162,"POWER":"ON","Wifi":{"AP":1,"SSId":"Cyberdyne `Systems","RSSI":62,"APMac":"9C:C7:A6:49:BB:31"}}
`

ascillato, 2018-05-01:
Hi,

I tried to reproduce the issue but my devices are working for hours without that behaviour. So, we have to do some tests to find out what is going on with yours.

Please, could you type on the console `weblog 4` and then post here the logging information when that 'stop receiving' happens? On the console you will see the KNX commands going and coming to the addresses registered.

Do you have another sonoff device? So as to add also a device to device communication to see if the devices can continue with the communication when this happens?

device to device could be:

DEVICE 1:
     BUTTON1 -> 1.1.1
     1.1.2 -> RELAY1

DEVICE 2:
     RELAY1 -> 1.1.2
     1.1.1 -> RELAY1 TOGGLE

ascillato, 2018-05-03:
ok,

Up to now we know for sure that
 - From KNX Network to your knx router, the communication is ok.
 - The Wifi signal is OK
 - The Tasmota can send and receive KNX telegrams for about 15 mins.
 - The Tasmota firmware version and config are OK

So, the problem is in the communication wifi router <-> Tasmota.
So, to aisolate wich device is having trouble, let's try this:

In your phone, please install a multicast tester app like this one:

[image]

And with this we can see the telegrams but over the wifi. Please, tell me, if when the "stop receiving" happens, if you still can see the KNX telegrams on the app.

RocketSience, 2018-05-04:
Hi,
very good idea to use this app!
I´ve done the Test.
The KNX telegrams are still visible in this app when the problem occurs
[image]

After disconnecting the mains from the Sonoff device its working again, you can see the response
[image]

whats now?

ascillato, 2018-05-05:
Hi,

Thanks a lot for your patience and for all the tests.

Please, let's try another firmware version. Now in releases there is the [v5.13.1a](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases) with some bugs solved from Tasmota (not related to this issue). This is to test if the problem is related or not to lwIP 2.0. The last release is built with lwIP 1.4 due to better stability.

RocketSience, 2018-05-07:
Hi,
i have testes the new release, with the same result as before.
Maybe some not Sonoff related KNX-telegrams lead to an error in the knx-receive-stack, like an overflow?

RocketSience, 2018-05-07:
Hi,
thank you for your effort!
Its not that some single telegrams are lost, when the reception stops no telegrams are processed, i could send them 100 times an not one produces an reaction. What's strange is that the other direction always works without problems.

ascillato, 2018-05-08:
Hi,

Now in releases there are the firmwares:

* **sonoff_KNX-DE_TEST.bin** - German Ver - TEST for ESP8266 v2.4.1 lwIP 2 LowMem w/esp-knx-ip-asyncUDP
* **sonoff_KNX-EN_TEST.bin** - English Ver - TEST for ESP8266 v2.4.1 lwIP 2 LowMem w/esp-knx-ip-asyncUDP

These have the latest libraries. please do not use sleep command.

jeylites, 2018-05-09:
@ascillato 
What compiler are you using?

ascillato, 2018-05-09:
> I’m starting to see a pattern. It seem very few people has got issues
> compiling with Platform io
>
> I going to try and to see it fixes the issue

I don't know. Here rocketscience is using the same firmware that I have on
my devices.

RocketSience, 2018-05-09:
Sorry for the late feedback.
I´ve testet the "sonoff_KNX-DE_TEST.bin" but with the same result.
Over the GUI it´s responding always without problems.

ascillato, 2018-05-09:
if you type status 0 on the console, can you see:

19:13:20 RSL: STATUS2 = {"StatusFWR":{"Version":"**5.13.1a**","BuildDateTime":"2018-05-07T22:06:01","Boot":31,"Core":"**2_4_1**","SDK":"2.2.1(cfd48f3)"}}

ascillato, 2018-05-09:
mmmm,

this issue is hard. I can not reproduce it at home, and the new firmware you have is using a totally different multicast routines.

do you have access to the serial connection of the device? because the esp-knx-ip library has a debug info that can be activated (on compilation time) that outputs to the serial interface. to see if the ESP8266 is seeing the KNX telegrams.

If you dont have access, I will have to add those debug routines to tasmota to display on the console.

RocketSience, 2018-05-09:
i think i have accesss to ther serial connection, i would try this.

ascillato, 2018-05-09:
Great!!!

I have added to the releases page the firmware files:

* **sonoff_KNX-EN_DEBUG.bin** - English Ver - DEBUG
* **sonoff_KNX-DE_DEBUG.bin** - German Ver - DEBUG

These ones are the same than **sonoff_KNX-EN** and **sonoff_KNX-DE** but with Debug information over serial.

Please, on the console, type the command `seriallog 4`, so as to have all the information over the serial.

Thanks

RocketSience, 2018-05-09:
Hi,
here are the results.
As long as the receiving works all sorts of knx-telegrams are received and displayed.
if an matching telegramm is found it is decoded.....
but if the reception stopps no telegramms at all are received...
at 21:23:43 i have switched off over the wep-gui
`
LEN: 19
Got packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xD0 0x11 0x2 0x39 0xC 0x3 0x0 0x80 0x1 0x68
ST: 0x530
MT: 0x29
ADDI: 0x0
C1: 0xBC
C2: 0xD0
DT: 0x1
HC: 0x5
EFF: 0x0
Source: 0x11 0x2
Dest: 0x39 0xC
CT: 0x2
 0x80 0x1 0x68==
Testing: 0x3D 0x46
21:23:02 WIF: PrÃ¼fe Verbindung...
21:23:02 WIF: verbunden

LEN: 17
Got packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xB4 0xD0 0x11 0x16 0x14 0xA 0x1 0x0 0x0
ST: 0x530
MT: 0x29
ADDI: 0x0
C1: 0xB4
C2: 0xD0
DT: 0x1
HC: 0x5
EFF: 0x0
Source: 0x11 0x16
Dest: 0x14 0xA
CT: 0x0
 0x0==
Testing: 0x3D 0x46

LEN: 17
Got packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xD0 0x11 0x32 0x14 0xA 0x1 0x0 0x41
ST: 0x530
MT: 0x29
ADDI: 0x0
C1: 0xBC
C2: 0xD0
DT: 0x1
HC: 0x5
EFF: 0x0
Source: 0x11 0x32
Dest: 0x14 0xA
CT: 0x1
 0x41==
Testing: 0x3D 0x46

LEN: 17
Got packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xD0 0x11 0xFA 0x3D 0x46 0x1 0x0 0x80
ST: 0x530
MT: 0x29
ADDI: 0x0
C1: 0xBC
C2: 0xD0
DT: 0x1
HC: 0x5
EFF: 0x0
Source: 0x11 0xFA
Dest: 0x3D 0x46
CT: 0x2
 0x80==
Testing: 0x3D 0x46
Found match
21:23:12 KNX: Received from 7.5.70 Befehl Write: 0 zu Relay  1
Creating packet with len 17
Sending packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xE0 0x11 0x27 0x3D 0x47 0x1 0x0 0x80
21:23:12 KNX: Relay  1 = 0 sent to 7.5.71
21:23:12 RSL: RESULT = {"POWER":"OFF"}
21:23:12 RSL: POWER = OFF
21:23:13 CFG: in Flash gespeichert am F6, zÃ¤hlen 355, Bytes 2560

LEN: 19
Got packet: 0x6 0x10 0x5 0x30 0x0 0x13 0x29 0x0 0xBC 0xD0 0x11 0x36 0x3B 0x29 0x3 0x0 0x80 0x14 0x50
ST: 0x530
MT: 0x29
ADDI: 0x0
C1: 0xBC
C2: 0xD0
DT: 0x1
HC: 0x5
EFF: 0x0
Source: 0x11 0x36
Dest: 0x3B 0x29
CT: 0x2
 0x80 0x14 0x50==
Testing: 0x3D 0x46
21:23:22 WIF: PrÃ¼fe Verbindung...
21:23:22 WIF: verbunden
21:23:43 WIF: PrÃ¼fe Verbindung...
21:23:43 WIF: verbunden
Creating packet with len 17
Sending packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xE0 0x11 0x27 0x3D 0x47 0x1 0x0 0x81
21:23:54 KNX: Relay  1 = 1 sent to 7.5.71
21:23:54 RSL: RESULT = {"POWER":"ON"}
21:23:54 RSL: POWER = ON
21:23:55 CFG: in Flash gespeichert am F5, zÃ¤hlen 356, Bytes 2560
Creating packet with len 17
Sending packet: 0x6 0x10 0x5 0x30 0x0 0x11 0x29 0x0 0xBC 0xE0 0x11 0x27 0x3D 0x47 0x1 0x0 0x80
21:23:55 KNX: Relay  1 = 0 sent to 7.5.71
21:23:55 RSL: RESULT = {"POWER":"OFF"}
21:23:55 RSL: POWER = OFF
21:23:56 CFG: in Flash gespeichert am F4, zÃ¤hlen 357, Bytes 2560
21:24:03 WIF: PrÃ¼fe Verbindung...
21:24:03 WIF: verbunden
21:24:12 RSL: STATE = {"Time":"2018-05-09T21:24:12","Uptime":"0T00:05:41","Vcc":3.194,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxx","RSSI":54,"APMac":"9C:C7:A6:49:BB:31"}}
21:24:24 WIF: PrÃ¼fe Verbindung...
21:24:24 WIF: verbunden
21:24:44 WIF: PrÃ¼fe Verbindung...
21:24:44 WIF: verbunden
21:25:05 WIF: PrÃ¼fe Verbindung...
21:25:05 WIF: verbunden
21:25:25 WIF: PrÃ¼fe Verbindung...
21:25:25 WIF: verbunden
21:25:46 WIF: PrÃ¼fe Verbindung...
21:25:46 WIF: verbunden
21:26:06 WIF: PrÃ¼fe Verbindung...
21:26:06 WIF: verbunden
21:26:27 WIF: PrÃ¼fe Verbindung...
21:26:27 WIF: verbunden

`

ascillato, 2018-05-09:
Hi

I think I found the problem ! :smile: 

I could reproduce this issue. 

If the sonoff device get disconnected from the wifi router, when reconnecting to the same router it loose the multicast connection for receiving packets. That's why a reboot solve the issue because on boot it opens again the UDP.

I will release a patch for this.

So, The issue was not related to esp lib versions, configurations nor compilation.

For the disconnection tests, I just unplug my wifi router and plug it again.

So, then with the patched version you should not have problems, but you should look why you have those micro-disconnections from your router ( may be too many wifi devices ? )

There is a very interesting and technical video about wifi routers installation by **davidelang** at https://www.usenix.org/conference/lisa12/technical-sessions/presentation/lang_david_wireless

ascillato, 2018-05-09:
Thanks a lot again for all the tests you made. Thanks to all your help was possible to be found the root of the problem. :+1:

ascillato, 2018-05-10:
Hi,

On releases page were updated all the firmwares with this bug fixed.

Please, check it.

Thanks!

RocketSience, 2018-05-10:
Hi,
i have testet the new release, but im sorry to say that it does not solve the problem.
btw, when do you sleep?

`
06:52:45 CMD: status 0
06:52:45 RSL: empfangenes topic /status, Datengröße 1, Daten 0
06:52:45 RSL: Gruppe 0, Index 1, Befehl STATUS, Daten 0
06:52:45 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
06:52:45 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:07:41","StartupUTC":"2018-05-10T05:45:04","Sleep":0,"BootCount":36,"SaveCount":373,"SaveAddress":"F4000"}}
06:52:45 RSL: STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018.05.10 01:35:30","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
06:52:45 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["xxxx",""],"TelePeriod":300,"SetOption":["02008001","55818000"]}}
06:52:45 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
06:52:45 RSL: STATUS5 = {"StatusNET":{"Hostname":"SONOFF_001","IPAddress":"192.168.0.33","Gateway":"192.168.0.253","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.253","Mac":"EC:FA:BC:10:56:38","Webserver":2,"WifiConfig":3}}
06:52:45 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu May 10 05:52:45 2018","Local":"Thu May 10 06:52:45 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:15","Sunset":"20:18"}}
06:52:45 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-05-10T06:52:45"}}
06:52:45 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-05-10T06:52:45","Uptime":"0T00:07:41","Vcc":3.172,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","RSSI":72,"APMac":"9C:C7:A6:49:BB:31"}}}
06:52:46 KNX: Start
06:52:47 KNX: Start
06:52:48 KNX: Start
06:52:49 KNX: Start
06:52:50 KNX: Start
06:52:51 WIF: Prüfe Verbindung...
06:52:51 WIF: verbunden
06:52:51 KNX: Start
06:52:52 KNX: Start
06:52:53 KNX: Start
`

ascillato, 2018-05-10:
ups... sorry. uploaded wrong file.

Fixed and updated files on release page.

Now, when there is a reconnection you will see _KNX Start_ on the console.

ascillato, 2018-05-10:
The **serial log disabled** message is always there. Now you see it because you have `weblog 4`. In the code, the _disable log_ does nothing to KNX.

Let's try this. I added the KNX_START routine that register on the router the multicast channel to receive, also on the SAVE KNX SETTINGS Button.

Please, when the disconnection happens, go to KNX Menu and press save.

After that try again to see if it receives telegrams. (everything without reboot)

If this works, we just need to find a way to make tasmota realize of this micro-disconnections to re-establish the UDP channel.

The new firmware for this testing is sonoff_KNX-DE_TEST.bin

ascillato, 2018-05-10:
Hi,

Do you have access to your wifi router configuration?

The configuration that is working fine for me is:

[image]

Note the **Multicast: IGMP v3**

ascillato, 2018-05-10:
This multcast disconnection seems to be a problem on the ESP8266 Libraries as explained on https://github.com/esp8266/Arduino/issues/ 1165.

So, the ESP8266 library have problems with some routers if IGMP snoop is enabled.

That explains why I can not reproduce this issue, and you can.

We tested the v2.3.0 and also v2.4.1 with lwIP 1.4 and 2.0 and all the time the fail is the same.

Let's try:
 * TEST firmware to see if a reconnection will work
 * Try to change the wifi router configuration to use **Multicast: IGMP v3** and **IGMP snoop: disabled**

RocketSience, 2018-05-10:
Hi,
first i tested only the new firmware, but the reconnection ("KNX: Start") does not help.
Then i changed my network hardware configuration, but there are no options like "IGMP Snoop".

I optimized the settings in the Fritzbox according to this site [https://avm.de/service/fritzbox/fritzbox-7390/wissensdatenbank/publication/show/592_Stoerungen-beim-Fernsehen-ueber-IPTV-Anschluss/](url)

If i disable the optimizing for IPTV (witch is IGMP-Snoop ON i think) the sonoff responses get very slow and most of the KNX-Telegrams to the sonoff get lost.

I think the esp should handle these disconnections, as they always can occour.
Resetting each device if my WLAN-Router needs an reboot is not the best option.

Actualy i think i better use the mqtt- or http-way and not the direct way over multicast, since this is not working stable in my hardware-setup

This is my network config:
**Before:** 
WLAN:
Fritzbox 7490 (Wlan)-> Wlan-Repeater -> Sonoff 
LAN:
Fritzbox 7490 (WLAN) -> D-Link DGS-1210-10P -> KNX-Router and KNX-Telegram-Source (EDOMI; ETS)

**Now:**
WLAN:
Fritzbox 7490 (Wlan)-> Wlan-Repeater -> Sonoff 
LAN:
Fritzbox 7490 (same box) -> KNX-Router and KNX-telegram-source (EDOMI; ETS; ...)

ascillato, 2018-05-10:
> i tested only the new firmware, but the reconnection ("KNX: Start") does not help.

mmm, very weird. The routine behind **KNX: Start** opens a new socket for multicast reception, meaning that it joins again the multicast group for the KNX IP:PORT Address. So, the repeater and the esp are not understanding each other. **If you connects direct to the Fritzbox, the "stop receiving" still happens?**

Sometimes the WLAN Repeaters have problems with some devices only. In your case your phone did not loose the multicast channel but the ESP does. It is known that the ESP has some issues with some routers/repeaters but the libraries are still on development to solve those.

> I think the esp should handle these disconnections, as they always can occour.

Agree, but we reach an issue beyond Tasmota KNX Driver and beyond Tasmota project. It is on ESP8266 Wifi Libraries. I think, you should post this issue on https://github.com/esp8266/Arduino because it is a problem with IGMP handshake between ESP and your Repeater in particular.

> Actualy i think i better use the mqtt- or http-way and not the direct way over multicast, since this is not working stable in my hardware-setup

Using MQTT you have a direct TCP connection that it is more reliable than multicast, and you can add also a home automation system like Home Assistant.

Future Ideas:
To increase reliability over KNX multicast I'm working on an different approach for Tasmota to Tasmota communication that is sending an ACK telegram when reception was OK, if not repeat telegram. But this will work just between Tasmotas and any KNX device that you can configure a behaviour like this.
Besides this, there is a KNX-IP Tunneling protocol option that it is very reliable but just few connections can be handled by the KNX router, meaning that you can have Tasmota to Tasmota with ACK communication and one Tasmota to broadcast everything to KNX-IP Tunnel.

ascillato, 2018-05-10:
Well, with all this, we found and fix a bug when resetting the wifi router the KNX needed to reconnect.

But we could not solve the issue with your repeater and the ESP IGMP handshake.

ascillato, 2018-05-14:
Closing the Issue.

I'm willing to help if you want to come again to use KNX when the reliability improvement is finished.

Again, Thanks a lot for your help! :+1:

closed 2018-05-14 by ascillato

ascillato, 2018-05-14:
Hi. Please take a look to issue 2643 from arendst/Sonoff-Tasmota.

That was a very similar issue than this and the problem was the device acting as repeater. 

Could you test if connecting Tasmota direct to your main router the issue resolves? 

Thanks