ascillato/Tasmota_KNX #90: KNX IP (Change IP of KNX Broadcast)

ascillato, 2018-05-06:
Add option to Change IP of KNX Broadcast. #2631 from Tasmota

ascillato, 2018-05-12:
I realize that up to now, there is no use case that need to use that option. Closing.

closed 2018-05-12 by ascillato

TheInfernoMan, 2018-05-13:
It is sad to hear this, but thanks for your time and your work on Sonoff Tasmota KNX 😥

ascillato, 2018-05-13:
Hi, I look for the KNX insideControl Gateway and it is a KNX certified gateway. So, if your model support multicast communication it is on the default address. So, please check the config of the gateway to see if it is turned on the multicast support. Also checks your wifi router if it has multicast turned on.

For the insidecontrol it is not needed to change the default broadcast address.

ascillato, 2018-05-13:
Please, remember that your IP Gateway Address has nothing to do with the Multicast IP Address.

IP and Multicast IP are totally different things. An IP is the address of a particular device on a network.

A multicast IP is a virtual address to where all the devices can talk to each other.

That's why, this option is not needed by now. Adding it will increase firmware size, making more difficult to add other options like ETS programming and so on.

If there is a use case that needs this, I can add it.

If you want I can make a firmware just for you with your modified Multicast Address just to test, but I think that your gateway will not recognize this new address due to is a KNX certified Gateway.

TheInfernoMan, 2018-05-13:
Ok thanks for your help, I will check if the ETS Application of the InSideControl gives me the option to enable Multicast. Thanks in advance :)

ascillato, 2018-05-13:
Look for your model also, Not all support multicast. Some just support KNXnet/IP Tunneling for ETS configuration.

ascillato, 2018-05-13:
check also your wifi router :+1:

ascillato, 2018-05-15:
Hi,

I made a Firmware ([sonoff_KNX-EN_TEST.bin](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.13.1b)) for you with the Multicast Address at 192.168.178.40 : 3671

It is in English and also has KNX_DEBUG_ON for the serial. If you have serial connector you should see there all the KNX telegrams that arrives and that are sent.

Please, test it to see if it works with that address.

Luis567, 2018-11-12:
Hello,

i have a short question also to this topic. When i have only a KNX IP Interface (Tunneling) and not a IP Router (which can multicast), is it then also possible to use sonoff tasmota knx gateway or only when the knx gateway can multicast?

The IP interface has a static ip adress in my network. The port for which is using in the interface is also 3671. 
Is there an option to add IP Tunneling to the firmware, so that it also work with a IP Interface? My IP Interface can for example 5 parallel tunneling connections.

ascillato, 2018-11-12:
Sonoff Tasmota KNX only supports KNX-IP multicast. Tunneling is not supported. Sorry

lassos, 2019-10-13:
Could you please add  tunneling support ? I just have a mdt knx ip interface without router function. so there is no way to send udp via multicast to 224.0.23.12:3671