ascillato/Tasmota_KNX #138: How to us temperatur of DS18B20 in KNX area?

misc2000, 2018-06-12:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff Basic with DS18B20 on GPIO14
- [X] Provide the output of command ``status 0`` :
STATUS 0 OUTPUT HERE
```can only provide later today if needed

Hello I have KNX all over my house and I've tried the current version (5.14.0b) to see how the new KNX integration is working. I tested with Sonoff Basic using platformIO on my MAC.

With the built-in switch and LED of a Sonoff Basic the KNX part is working perfect.

But now I need a temperature measurement for a current project.
For this I have connected a 1-wire temperature sensor (DS18B20) with the needed resistor to the Sonoff Basic to GPIO14 and configure it in the Tasmota device config to the appropriate GPIO.
After a reboot, I also get the name of the sensor and the current temperature on the Tasmota start webpage of the device.

BUT
In the KNX configuration page, I can not select the temperature sensor as "Source / Sensor" for sending to group addresses. I only see this in the pulldown menu: relay, button, KNX TX 1, ... KNX TX 5)
Has anyone else tried this successfully, do I have to change anything else in the source code / sketch to be able to use the sensor also in the KNX area?
Or is this a bug?
Michael

misc2000, 2018-06-12:
yes but only late afternoon

ascillato, 2018-06-12:
Nevermind. Found the issue. I have never included the support for that sensor in knx. Will include for next release. Thanks a lot for finding this. :+1:

misc2000, 2018-06-12:
OK or not ok because I just bought 5 of this sensors because I thought that all sensors of this list are also working in the KNX side
https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration
Is there anywere a list where I can see what sensors are supported?
And can you also implement to support up to 5 of this 1-wire sensors.
If any testing is needed during implemetation I can help...

ascillato, 2018-06-12:
Hi,

I have just add the support for DS18B20 in the KNX Driver for Tasmota.

I don't have a DS18B20 to test. So, can you test that please? I updated the code in [Tasmota dev branch](https://github.com/arendst/Sonoff-Tasmota), in this [KNX dev branch](https://github.com/ascillato/Sonoff-Tasmota_KNX) and also I added the precompiled bins with this upgrade in this [Sonoff-Tasmota_KNX repository](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases)

Up to now, Tasmota supports only one DS18B20 sensor to be used in KNX. For supporting more sensors of the same type, a major modification would be needed in the KNX Menu and in the KNX driver. I understand that it is very useful, but by now, there are no plans to work on adding that feature soon.

The implemented features in KNX for Tasmota are:

General:
* buttons (just push)
* relays (on/off/toggle)
* lights (led strips, etc. but just on/off)

Sensor lists that you can use in KNX is (only one sensor per type):
* Temperature
* Humidity
* Energy (v, i, power)

For using rules:
* send KNX command (on/off)
* receive KNX command (on/off)
* send values by KNX (any float type, temperature for example)
* receive a KNX read request

The Features which I'm working on at this moment:
* For using rules: receive values (any float type, temperature for example)
* Optimizing KNX Driver code for better RAM and FLASH usage.

Thanks again to point all this out. :+1: . I hope I can update soon the wiki with this information.

Please, confirm if the DS18B20 works when you have some time. Thanks.

ascillato, 2018-06-12:
Added this info to the Tasmota Wiki !

misc2000, 2018-06-12:
Thanks, that was fast. I will start testing right now...

misc2000, 2018-06-12:
sorry for the delay but I am very new to Arduino and PlatformIO. I have problems to compile it. And I am not able to upload the .bin with the Tasmota OTA-Update (to big) Is there any fast way to upload your precompiled bin with PlatformIO (on the MAC) over the USB Adapter?

ascillato, 2018-06-12:
I provided you the precompiled also. By OTA, use the minimal first. By wire use the esptool.py

ascillato, 2018-06-12:
**Download Firmware and Flash**

You can use the precompiled binaries v6.0.0a from:
https://github.com/ascillato/Sonoff-Tasmota_KNX/releases

These can be uploaded using **OTA** or using the tool **esptool.py**. Information at: https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

( If using **_OTA_**, upload first the **minimal** version and after that, the **final** version you want )
( If using **_esptool_**, upload **directly** the final version you want )

**_OR_**

**Download Source Code, Compile and Flash**

You can download the source code from:
https://github.com/ascillato/Sonoff-Tasmota_KNX or from https://github.com/arendst/Sonoff-Tasmota

And compile and flash with Arduino or Platformio. Tutorials at: https://github.com/arendst/Sonoff-Tasmota/wiki/Tutorials

Hope this helps.

misc2000, 2018-06-12:
after remember some basic possible errors to update a sonoff Basic and find a lose wire I managed to update.
But first test is only half good.
I can configure now the temperature but if I read the KNX group address I only get the value 00h from the device :-(

ascillato, 2018-06-12:
The temperature value is updated every teleperiod only. If you wait until your teleperiod, it stills give you zero?

misc2000, 2018-06-12:
you are right. The default teleperiod is 5 min. And it looks like in the first 5 min I good the Zero. Now get the right value after the first push was send also on the read request.

misc2000, 2018-06-12:
But it would be nice if you can load the right value direct to the KNX side that a read request have the right value or if the first teleperiod just starts once after a reconfigure or restart of the device...

ascillato, 2018-06-12:
After a restart, you have a teleperiod after Tasmota reads all the sensors. May be you have restarted the device and ask for temperature before Tasmota finish communicating and reading all sensors.

Please, do a restart and try again after about 10 seconds.

misc2000, 2018-06-12:
after a restart there is no value be pushed to the KNX GA and also my reads after 10, 20, 30,... seconds give me all the time 00h until the 5 min default teleperiod is over and the first temperature is pushed to the GA. After that I also get the right value to the read requests

ascillato, 2018-06-12:
Ok. Nice that it is working now.

The teleperiod and the sensor readings are outside the knx driver. Give me some time. I will investigate that and then I will post an issue in Tasmota so Theo can check that.
Seems to me that to be solved will take some time. Please be patient.

ascillato, 2018-06-13:
Hi,

A question. Do you have a MQTT broker configured? If not, that is the issue with the initial teleperiod.

I found a workaround for making Tasmota have the Temperature value available from Boot Time for KNX.

* Please, go to the console.

* Turn on rules: `rule 1`

* Add the rule: `rule on System#Boot do teleperiod 300 endon`

This rule will trigger a teleperiod once the device has booted.

misc2000, 2018-06-13:
yes could be the issue. I don't have a MQTT setup at all. I will verify the workaround this afternoon.

misc2000, 2018-06-13:
hi, 
your workaround is not working because the trigger of `System#Boot` is (from wiki:) after mqtt is initialized and I do not us mqtt. (Maybe a other enhancement to make this independent from mqtt)
But I found `Time#Initialized` and with this rule it works
`rule on System#Boot do teleperiod 300 endon` 
with this rule I got a pushed temperature value to the KNX GA direct after the Sonoff reboot and also direct right temperature values when I send read requests to the KNX group address.

ascillato, 2018-06-13:
Great! solved!

Thanks a lot for finding these issues and testing :+1: 

So the rule should be `rule on Time#Initialized do teleperiod 300 endon`, right?

I will try to make that the KNX driver force a sensor reading at startup if there is no MQTT.

misc2000, 2018-06-14:
right. should be `rule on Time#Initialized do teleperiod 300 endon` 
I also update the copy past error in my last rule post.
Let me know if you have any improvements to test.
Do you see also any workaround (maybe with some static mods in the sketch) to support more than one 1-wire sensor?

misc2000, 2018-06-14:
One idea,
should it be a workaround to connect each DS18B20 to a own GPIO ?

ascillato, 2018-06-14:
Hi,

Tasmota does support more than one DS18B20 but only for MQTT. For KNX, a major modification is needed.

If you have multiple DS18B20 sensors in the same GPIO, can you provide the output in the console for the tele message? May be, depending on how the JSON is, can be done a workaround with rules.

misc2000, 2018-06-14:
right now are 2 DS18B20 connected to the same GPIO:
```
20:17:23 RSL: tele/sonoff/STATE = {"Time":"2018-06-14T20:17:23","Uptime":"0T00:55:09","Vcc":3.466,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MISC-NET","RSSI":100,"APMac":"38:10:D5:84:68:4B"}}
20:17:23 KNX: Temperatur sent to 6.0.0 
20:17:23 RSL: tele/sonoff/SENSOR = {"Time":"2018-06-14T20:17:23","DS18B20-1":{"Id":"0517B00B4DFF","Temperature":30.8},"DS18B20-2":{"Id":"0517B01EE1FF","Temperature":23.7},"TempUnit":"C"}
```

misc2000, 2018-06-15:
hi I give it a try but don't work.
I did assign a GA (0/0/22)to KNX TX1 and GA 6/0/0 is used for the temperature of sensor 1.
And use this rule:

> rule on tele-DS18B20-2#temperature do var1 %value%;knxtx_val1 %var1% endon
but nothing is send to the assigned GA
Console log looks like this:
```
17:06:26 RSL: tele/sonoff/STATE = {"Time":"2018-06-15T17:06:26","Uptime":"0T00:10:00","Vcc":3.468,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MISC-NET","RSSI":100,"APMac":"38:10:D5:84:68:4B"}}
17:06:26 KNX: Temperatur sent to 6.0.0 
17:06:26 RSL: tele/sonoff/SENSOR = {"Time":"2018-06-15T17:06:26","DS18B20-1":{"Id":"0517B00B4DFF","Temperature":23.6},"DS18B20-2":{"Id":"0517B01EE1FF","Temperature":23.4},"TempUnit":"C"}
17:06:26 RUL: TELE-DS18B20-2#TEMPERATURE performs "var1 23.4;knxtx_val1 23.4;knxt"
17:06:26 RSL: stat/sonoff/RESULT = {"Var1":"23.4;knxt"}
```

misc2000, 2018-06-15:
just to verify that the GA is working I did send the DO part of the rule (copied from the console log) as command into the console `knxtx_val1 23.4` and the value is send to the bus. 
Therefor it looks like my rule is ok designed but not performed :-(

misc2000, 2018-06-15:
with putting knxtx_val1 in first place of the do part the value is send to the bus.
But I direct found the next problem the temperature value will be only send without the position after decimal point
see the log:
```
18:14:02 CMD: rule on tele-DS18B20-2#temperature do knxtx_val1 %value%;var1 %value% endon on event#knxrx_req1 do knxtx_val1 %val1%
18:14:02 RSL: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","Free":400,"Rules":"on tele-DS18B20-2#temperature do knxtx_val1 %value%;var1 %value% endon on event#knxrx_req1 do knxtx_val1 %val1%"}
18:14:08 MQT: Verbindungsversuch...
18:14:19 MQT: Verbindungsversuch...
18:14:27 RSL: tele/sonoff/STATE = {"Time":"2018-06-15T18:14:27","Uptime":"0T01:18:01","Vcc":3.466,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MISC-NET","RSSI":100,"APMac":"38:10:D5:84:68:4B"}}
18:14:27 KNX: Temperatur sent to 6.0.0 
18:14:27 RSL: tele/sonoff/SENSOR = {"Time":"2018-06-15T18:14:27","DS18B20-1":{"Id":"0517B00B4DFF","Temperature":23.7},"DS18B20-2":{"Id":"0517B01EE1FF","Temperature":23.6},"TempUnit":"C"}
18:14:27 RUL: TELE-DS18B20-2#TEMPERATURE performs "knxtx_val1 23.6;var1 23.6"
18:14:27 KNX: KNX TX 1 = 23 sent to 6.0.22
18:14:27 RSL: stat/sonoff/RESULT = {"KnxTx_Val1":"23"}
```

ascillato, 2018-06-18:
Hi,

Thanks a lot for the testing.

Will investigate your findings

ascillato, 2018-06-18:
your rule `rule on tele-DS18B20-2#temperature do knxtx_val1 %value%;var1 %value% endon on event#knxrx_req1 do knxtx_val1 %var1%`

is missing the last `endon`.

should be:

 `rule on tele-DS18B20-2#temperature do knxtx_val1 %value%;var1 %value% endon on event#knxrx_req1 do knxtx_val1 %var1% endon`

misc2000, 2018-06-19:
The rule is working right now. :-)
But could it be that the firmware have some long time running issues?

I have now 2 units running since last Sunday with your new version 6.x:
A) both do not show any web interface anymore after 30 hours. 
But still send the temperature values every 2 minutes to the KNX bus. 
And I can also ping them all the time. Now tonight the web-interface of both are working again.
 
B) one unit did today not answer to any KNX read requests anymore at all. 
After a soft reboot from the web-interface the read requests are answered again without any configuration changes.

And can you please take a look about the missing digits after decimal point I reported some days ago.
Thanks

ascillato, 2018-06-20:
Hi, will look into it. I was very busy. please be patient. thanks

ascillato, 2018-06-26:
Hi, I found the issue with the decimal values. Tasmota internally only sends integers to drivers. I will fix it soon.

About your long time running issues, I don't know what is happening there. Mine are working ok. May be an issue on your wifi?

ascillato, 2018-06-26:
Hi, your finding about the decimal values is also in the ADD, SUB and MULT commands. Will try to add decimals to all those commands too. Thanks again for your tests! :+1:

ascillato, 2018-06-26:
> could it be that the firmware have some long time running issues

I recall that some people had that type of issue in Tasmota. Don't know the root of their issue but a workaround they did is to automatically reboot Tasmota once per day using a timer and a rule. May be this help you too as a workaround.

misc2000, 2018-06-26:
good to hear that you found the source of the decimal problem because without it its more or less useless for me. Let me know if I should make some tests. And thanks for the hint with the daily reboot, I will give it a try...

ascillato, 2018-06-27:
I made the https://github.com/arendst/Sonoff-Tasmota/issues/3083 for tracking the decimal truncation bug.

So, if your issue is solved, let's close this one. Please, feel free to reopen it or open a new one if you need. Thanks.

closed 2018-06-27 by ascillato

ascillato, 2018-06-28:
Now Tasmota supports decimal values for KNXTX_VALx command and also for Arithmetic commands for rules.

Also the precompiled binaries for KNX were [released](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v6.0.0b)

misc2000, 2018-06-28:
I will test it tomorrow...

misc2000, 2018-07-01:
I managed to compile with the multi DS18B20 support and can confirm that decimal values for KNXTX_VALx work.

ascillato, 2018-07-01:
Great! Thanks :+1: