ascillato/Tasmota_KNX #140: SR04 KNX support

Winni66, 2018-06-13:
I have ordered SR04 sensor, but I think there is no possibility to use it together with KNX yet, am I wrong?
Would be great if this feature would be integrated.

ascillato, 2018-06-13:
By now, there is no direct way to configure that from the KNX menu.

What you can do to use the SR04 now, is using rules and the `KnxTx_val1` command. There is an example of this in the Tasmota wiki.

I will add the support to the SR04 but not soon.

BTW, can you share in what scenario you are going to use the SR04 ?

Winni66, 2018-06-13:
I want to control the volume of my water reservoir in the garden (level in cm)

Winni66, 2018-06-13:
In addition I think about checking whether my car is in the garage or not

ascillato, 2018-06-13:
Cool! :+1:

ascillato, 2018-06-28:
Hi,

Until the support by the KNX menu is done, you can try the following so as to use SR04 sensor in KNX:

```
rule on tele-SR04#distance do knxtx_val1 %value% endon
```

Remember that you will have to uncomment [the line 314 in user_config.h file](https://github.com/ascillato/Sonoff-Tasmota_KNX/blob/development/sonoff/user_config.h#L314) and compile Tasmota from source, because it is not included the SR04 library in the precompiled binaries I provide in the Release page. Sorry. This is explained in the release page.

ascillato, 2018-06-28:
Hi,

I'm running some memory issues with the KNX Menu and I have to optimize the code, so I will have to dismiss adding an extra option to the KNX Menu. Sorry.

But I will add for next release the SR04 Library, so anyone that want to use this sensor in KNX, will be able to do it by just adding the above rule.

Sorry again.

Thanks a lot for all your findings, testings and patience! :+1:

closed 2018-06-28 by ascillato

ascillato, 2018-06-28:
Now, it is enabled by default the SR04 Library in the source code.

ascillato, 2018-06-28:
Now Tasmota supports decimal values for KNXTX_VALx command and also for Arithmetic commands for rules.

Also the precompiled binaries for KNX were [released](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v6.0.0b)