ascillato/Tasmota_KNX #141: Using rules for updating KNX values

Winni66, 2018-06-14:
I'm using Tasmota 6.0a from here with TH16. I want to get automatic value update for KNX when value has change more than 2 unit. I'm using proposed rule.

Maybe there is a problem with publish command?

21:31:53 CMD: rule on SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %var1%; var2 %var1%; add1 2; sub2 2 endon on SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %var2%; var1 %var2%; add1 2; sub2 2 endon on SI7021#humidity>%var3% do backlog var3 %value%; publish stat/sonoff/humidity %var3%; var4 %var3%; add3 2; sub4 2 endon on SI7021#humidity<%var4% do backlog var4 %value%; publish stat/sonoff/humidity %var4%; var3 %var4%; add3 2; sub4 2 endon
21:31:53 SRC: WebConsole from 192.168.5.9
21:31:53 RSL: empfangenes topic /rule, Datengröße 485, Daten on SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %var1%; var2 %var1%; add1 2; sub2 2 endon on SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %var2%; var1 %var2%; add1 2; sub2 2 endon on SI7021#humidity>%var3% do backlog var3 %value%; publish stat/sonoff/humidity %var3%; var4 %var3%; add3 2; sub4 2 endon on SI7021#humidity<%var4% do backlog var4 %value%; publish stat/sonoff/humidity %var4%; var3 
21:31:53 RSL: Gruppe 0, Index 1, Befehl RULE, Daten on SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %var1%; var2 %var1%; add1 2; sub2 2 endon on SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %var2%; var1 %var2%; add1 2; sub2 2 endon on SI7021#humidity>%var3% do backlog var3 %value%; publish stat/sonoff/humidity %var3%; var4 %var3%; add3 2; sub4 2 endon on SI7021#humidity<%var4% do backlog var4 %value%; publish stat/sonoff/humidity %var4%; var3 %var4%; add
21:31:53 RSL: RESULT = {"Rule1":"ON","Once":"OFF","Free":26,"Rules":"on SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %var1%; var2 %var1%; add1 2; sub2 2 endon on SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %var2%; var1 %var2%; add1 2; sub2 2 endon on SI7021#humidity>%var3% do backlog var3 %value%; publish stat/sonoff/humidity %var3%; var4 %var3%; add3 2; sub4 2 endon on SI7021#humidity<%var4% do backlog var4 %value%; publish stat/sonoff/humidity %va ...
21:31:53 CFG: in Flash gespeichert am FA, zählen 85, Bytes 3584
21:31:55 DHT: erhalten 01, BF, 01, 06, C7 =? C7
21:32:14 DHT: erhalten 01, BE, 01, 06, C6 =? C6
21:32:18 DHT: erhalten 01, BE, 01, 06, C6 =? C6
21:32:22 DHT: erhalten 01, CD, 01, 06, D5 =? D5
21:32:22 RUL: SI7021#HUMIDITY>%VAR3% performs "backlog var3 46.10; publish stat/sonoff/humidity 45; var4 45; add3 2; sub4 2"
21:32:22 SRC: Rule
21:32:22 RSL: empfangenes topic /backlog, Datengröße 68, Daten var3 46.10; publish stat/sonoff/humidity 45; var4 45; add3 2; sub4 2
21:32:22 RSL: Gruppe 0, Index 1, Befehl BACKLOG, Daten var3 46.10; publish stat/sonoff/humidity 45; var4 45; add3 2; sub4 2
21:32:22 SRC: Backlog
21:32:22 RSL: empfangenes topic /var3, Datengröße 5, Daten 46.10
21:32:22 RSL: Gruppe 0, Index 3, Befehl VAR, Daten 46.10
21:32:22 RSL: RESULT = {"Var3":"46.10"}
21:32:23 SRC: Backlog
21:32:23 RSL: empfangenes topic /publish, Datengröße 23, Daten stat/sonoff/humidity 45
21:32:23 RSL: Gruppe 0, Index 1, Befehl PUBLISH, Daten stat/sonoff/humidity 45
21:32:23 RSL: RESULT = {"Command":"Unknown"}
21:32:24 SRC: Backlog
21:32:24 RSL: empfangenes topic /var4, Datengröße 2, Daten 45
21:32:24 RSL: Gruppe 0, Index 4, Befehl VAR, Daten 45
21:32:24 RSL: RESULT = {"Var4":"45"}
21:32:24 SRC: Backlog
21:32:24 RSL: empfangenes topic /add3, Datengröße 1, Daten 2
21:32:24 RSL: Gruppe 0, Index 3, Befehl ADD, Daten 2
21:32:24 RSL: RESULT = {"Add3":"48"}
21:32:25 SRC: Backlog
21:32:25 RSL: empfangenes topic /sub4, Datengröße 1, Daten 2
21:32:25 RSL: Gruppe 0, Index 4, Befehl SUB, Daten 2
21:32:25 RSL: RESULT = {"Sub4":"43"}
21:32:25 DHT: erhalten 01, F6, 01, 07, FF =? FF
21:32:26 RUL: SI7021#HUMIDITY>%VAR3% performs "backlog var3 50.20; publish stat/sonoff/humidity 48; var4 48; add3 2; sub4 2"
21:32:26 SRC: Rule
21:32:26 RSL: empfangenes topic /backlog, Datengröße 68, Daten var3 50.20; publish stat/sonoff/humidity 48; var4 48; add3 2; sub4 2
21:32:26 RSL: Gruppe 0, Index 1, Befehl BACKLOG, Daten var3 50.20; publish stat/sonoff/humidity 48; var4 48; add3 2; sub4 2
21:32:26 SRC: Backlog
21:32:26 RSL: empfangenes topic /var3, Datengröße 5, Daten 50.20
21:32:26 RSL: Gruppe 0, Index 3, Befehl VAR, Daten 50.20
21:32:26 RSL: RESULT = {"Var3":"50.20"}
21:32:27 SRC: Backlog
21:32:27 RSL: empfangenes topic /publish, Datengröße 23, Daten stat/sonoff/humidity 48
21:32:27 RSL: Gruppe 0, Index 1, Befehl PUBLISH, Daten stat/sonoff/humidity 48
21:32:27 RSL: RESULT = {"Command":"Unknown"}
21:32:27 SRC: Backlog
21:32:27 RSL: empfangenes topic /var4, Datengröße 2, Daten 48
21:32:27 RSL: Gruppe 0, Index 4, Befehl VAR, Daten 48
21:32:27 RSL: RESULT = {"Var4":"48"}

One question in addtion: I wanted to give acces to TH16 with port forwarding 10080 -> TH16:80 but console window is empty, same for area with values above menu-buttons. Is there anything I can do to avoid this?

ascillato, 2018-06-18:
Hi

> I'm using Tasmota 6.0a from here with TH16. I want to get automatic value update for KNX when value has change more than 2 unit. I'm using proposed rule.
> Maybe there is a problem with publish command?

will investigate

> One question in addtion: I wanted to give acces to TH16 with port forwarding 10080 -> TH16:80 but console window is empty, same for area with values above menu-buttons. Is there anything I can do to avoid this?

Tasmota is not prepared to be port forwaded to internet. I think it is not easy to make it work and also it is not safe for your device. Tasmota is prepared for being used in a home automation software (that's why it has MQTT as a core), and your home automation software can be safely exposed to internet. I recommend HomeAssistant as it is very complete and also supports KNX.

ascillato, 2018-06-26:
Hi,

I have tested your rule and I did few changes to make it work:

for example, for temperature:

`rule on tele-SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %value%; var2 %value%; add1 2; sub2 2 endon on tele-SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %value%; var1 %value%; add1 2; sub2 2 endon`

You can use `tele-` or not. That depends how often you want the updates.

Please, test it. Thanks.

Winni66, 2018-06-27:
Sorry, but it seems I'm still to stupid for that. Rule is triggered, but no value sent

21:50:31 SRC: Backlog
21:50:31 RSL: empfangenes topic /var1, Datengröße 5, Daten 26.10
21:50:31 RSL: Gruppe 0, Index 1, Befehl VAR, Daten 26.10
21:50:31 RSL: RESULT = {"Var1":"26.10"}
21:50:31 SRC: Backlog
21:50:31 RSL: empfangenes topic /publish, Datengröße 22, Daten stat/sonoff/temp 26.10
21:50:31 RSL: Gruppe 0, Index 1, Befehl PUBLISH, Daten stat/sonoff/temp 26.10
21:50:31 RSL: RESULT = {"Command":"Unknown"}
21:50:32 SRC: Backlog
21:50:32 RSL: empfangenes topic /var2, Datengröße 5, Daten 26.10
21:50:32 RSL: Gruppe 0, Index 2, Befehl VAR, Daten 26.10
21:50:32 RSL: RESULT = {"Var2":"26.10"}
21:50:33 SRC: Backlog
21:50:33 RSL: empfangenes topic /add1, Datengröße 1, Daten 1
21:50:33 RSL: Gruppe 0, Index 1, Befehl ADD, Daten 1
21:50:33 RSL: RESULT = {"Add1":"27"}
21:50:33 SRC: Backlog
21:50:33 RSL: empfangenes topic /sub2, Datengröße 1, Daten 1
21:50:33 RSL: Gruppe 0, Index 2, Befehl SUB, Daten 1
21:50:33 RSL: RESULT = {"Sub2":"25"}
21:50:34 DHT: erhalten 01, 9F, 01, 04, A5 =? A5
21:50:37 DHT: erhalten 01, A0, 01, 04, A6 =? A6

ascillato, 2018-06-27:
Hi, 

Seems that the command `Publish `is not recognized. Weird. Please try the command alone:
`publish stat/sonoff/temp 26.10`
to see what happens

ascillato, 2018-06-27:
Ah, remember that `publish `is for send data over **MQTT** not for KNX

Winni66, 2018-06-27:
I thought your example is for KNX because you explained that sending on changes in addition to timeschedule is possible with rules, so is there a misunderstanding or do I have just to change the syntax? I couldn't find an updated wiki for this.

ascillato, 2018-06-27:
A rules example but for sending data over KNX could be:

```
rule on tele-SI7021#temperature>%var1% do backlog var1 %value%; knxtx_val1 %value%; var2 %value%; add1 2; sub2 2 endon on tele-SI7021#temperature<%var2% do backlog var2 %value%; knxtx_val1 %value%; var1 %value%; add1 2; sub2 2 endon
```

This rule should send data over KNX to the group address set in the KNX menu as KNX_TX1

Winni66, 2018-06-27:
Great, it becomes better and better....
Now I have the problem that it seems that all the variables are integer, therefore the sent value is integer and also the differences can be only full degrees.
Perhaps you can help once more ;-)

ascillato, 2018-06-27:
Great!

I will add decimals for that. Hope to be soon. :+1:

Winni66, 2018-06-27:
Next issue:  I have added humidity rule as rule2 and used knxtx_val2. Did right setup for humidity GA with KNX TX 2, but the value is sent to temperature GA (perhaps same issue as it was for request?) Would you please be so kind an check.

ascillato, 2018-06-27:
will investigate

Winni66, 2018-06-27:
Ups, forget  about my last comment, changed val1 only once to val2, 

It's working fine also for humidity

ascillato, 2018-06-27:
Great!!!!! :+1:

ascillato, 2018-06-27:
I made the https://github.com/arendst/Sonoff-Tasmota/issues/3083 for tracking the decimal truncation bug.

So, if your question is solved, let's close this issue. Please, feel free to reopen it or open a new one if you need. Thanks.

ascillato, 2018-06-28:
Now Tasmota supports decimal values for KNXTX_VALx command and also for Arithmetic commands for rules.

Also the precompiled binaries for KNX were [released](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v6.0.0b)

Winni66, 2020-04-16:
I will reactivate this issue, because I'm struggeling with Gosund power values. I have power-values which are changing all the time and I wanted to limit KNX telegrams. I did the following:

on system#boot do backlog var1 0; var2 0 endon on Energy#Power>%var1% do backlog var1 %value%; mult1 1.05; add1 2; knxtx_val1 %value%; var2 %value%; mult2 0.95; sub2 2 endon on Energy#Power<%var2% do backlog var2 %value%; mult2 0.95; sub2 2; knxtx_val1 %value%; var1 %value%; mult1 1.05; add1 2 endon

This should lead to sending value to KNX when value changes for at least 2 watt + 5 percent of actual value. But it doesn't work properly. Maybe the new value arrives before the rule is processed completely?

22:22:20 RUL: ENERGY#POWER>%VAR1% performs "backlog var1 254.400; mult1 1.05; add1 2; knxtx_val1 254.400; var2 254.400; mult2 0.95; sub2 2"
22:22:20 RSL: RESULT = {"Mult2":"224.358"}
22:22:21 RSL: RESULT = {"Sub2":"222.358"}
22:22:21 RSL: RESULT = {"Var1":"241.100"}
22:22:21 RSL: RESULT = {"Mult1":"253.155"}
22:22:21 RSL: RESULT = {"Add1":"255.155"}
22:22:22 KNX: KNX TX 1 = 241.10 sent to 8.6.50
22:22:22 RSL: RESULT = {"KnxTx_Val1":"241.10"}
22:22:22 RSL: RESULT = {"Var2":"241.100"}
22:22:22 RUL: ENERGY#POWER<%VAR2% performs "backlog var2 227.600; mult2 0.95; sub2 2; knxtx_val1 227.600; var1 227.600; mult1 1.05; add1 2"
22:22:22 RSL: RESULT = {"Mult2":"229.045"}
22:22:22 RSL: RESULT = {"Sub2":"227.045"}
22:22:23 RSL: RESULT = {"Var1":"254.400"}
22:22:23 RSL: RESULT = {"Mult1":"267.120"}
22:22:23 RSL: RESULT = {"Add1":"269.120"}
22:22:23 KNX: KNX TX 1 = 254.40 sent to 8.6.50
22:22:23 RSL: RESULT = {"KnxTx_Val1":"254.40"}
22:22:24 RSL: RESULT = {"Var2":"254.400"}
22:22:24 RUL: ENERGY#POWER<%VAR2% performs "backlog var2 228.900; mult2 0.95; sub2 2; knxtx_val1 228.900; var1 228.900; mult1 1.05; add1 2"
22:22:24 RSL: RESULT = {"Mult2":"241.680"}
22:22:24 RSL: RESULT = {"Sub2":"239.680"}
22:22:24 RUL: ENERGY#POWER<%VAR2% performs "backlog var2 227.700; mult2 0.95; sub2 2; knxtx_val1 227.700; var1 227.700; mult1 1.05; add1 2"
22:22:25 RSL: RESULT = {"Var2":"227.600"}
22:22:25 RSL: RESULT = {"Mult2":"216.220"}
22:22:25 RSL: RESULT = {"Sub2":"214.220"}
22:22:25 KNX: KNX TX 1 = 227.60 sent to 8.6.50
22:22:25 RSL: RESULT = {"KnxTx_Val1":"227.60"}
22:22:26 RSL: RESULT = {"Var1":"227.600"}
22:22:26 RUL: ENERGY#POWER>%VAR1% performs "backlog var1 228.000; mult1 1.05; add1 2; knxtx_val1 228.000; var2 228.000; mult2 0.95; sub2 2"
22:22:26 RSL: RESULT = {"Mult1":"238.980"}
22:22:26 RSL: RESULT = {"Add1":"240.980"}
22:22:26 RSL: RESULT = {"Var2":"228.900"}
22:22:26 RSL: RESULT = {"Mult2":"217.455"}
22:22:27 RSL: RESULT = {"Sub2":"215.455"}
22:22:27 KNX: KNX TX 1 = 228.90 sent to 8.6.50
22:22:27 RSL: RESULT = {"KnxTx_Val1":"228.90"}
22:22:27 RSL: RESULT = {"Var1":"228.900"}
22:22:27 RSL: RESULT = {"Mult1":"240.345"}
22:22:28 RSL: RESULT = {"Add1":"242.345"}
22:22:28 RSL: RESULT = {"Var2":"227.700"}
22:22:28 RSL: RESULT = {"Mult2":"216.315"}
22:22:28 RSL: RESULT = {"Sub2":"214.315"}
22:22:29 KNX: KNX TX 1 = 227.70 sent to 8.6.50
22:22:29 RSL: RESULT = {"KnxTx_Val1":"227.70"}
22:22:29 RSL: RESULT = {"Var1":"227.700"}
22:22:29 RSL: RESULT = {"Mult1":"239.085"}
22:22:29 RSL: RESULT = {"Add1":"241.085"}
22:22:29 RSL: RESULT = {"Var1":"228.000"}
22:22:30 RSL: RESULT = {"Mult1":"239.400"}
22:22:30 RSL: RESULT = {"Add1":"241.400"}
22:22:30 KNX: KNX TX 1 = 228.00 sent to 8.6.50
22:22:30 RSL: RESULT = {"KnxTx_Val1":"228.00"}
22:22:30 RSL: RESULT = {"Var2":"228.000"}
22:22:31 RSL: RESULT = {"Mult2":"216.600"}
22:22:31 RSL: RESULT = {"Sub2":"214.600"}

perhaps you can help?

ascillato, 2020-04-16:
Ok,

**_TIPs:_**
* Let's speed up `backlog` by using `SetOption34 10`
* Let's reduce few bytes your rules by deleting the space after the ; in the `backlog`. That space is supported but not needed.
* Let's reduce the amount of commands and steps by using `SCALE` and using `EVENT`

**_We can rewrite those rules like:_**

_Let's make sure that ONE SHOT is disabled in the rule by `rule1 4`_

```lua
SetOption34 10
rule1 4
rule1 1
rule1
  on system#boot do backlog var1 0;var2 0 endon
  on Energy#Power>%var1% do event threshold=%value% endon
  on Energy#Power<%var2% do event threshold=%value% endon
  on event#threshold do backlog knxtx_val1 %value%;SCALE1 %value%,0,100,0,105;SCALE2 %value%,0,100,0,95;add1 2;sub2 2 endon
```

**_All together for copy and paste in 2 steps:_**

```lua
backlog SetOption34 10;rule1 4;rule1 1
rule1 on system#boot do backlog var1 0;var2 0 endon on Energy#Power>%var1% do event threshold=%value% endon on Energy#Power<%var2% do event threshold=%value% endon on event#threshold do backlog knxtx_val1 %value%;SCALE1 %value%,0,100,0,105;SCALE2 %value%,0,100,0,95;add1 2;sub2 2 endon
```

**_Testing:_**

I tested these rules but by using `VAR3` instead of `ENERGY#POWER`:

```lua
rule1 on system#boot do backlog var1 0;var2 0 endon on var3#state>%var1% do event threshold=%value% endon on var3#state<%var2% do event threshold=%value% endon on event#threshold do backlog knxtx_val1 %value%;SCALE1 %value%,0,100,0,105;SCALE2 %value%,0,100,0,95;add1 2;sub2 2 endon
```
Console:
```lua
20:18:57 CMD: var3 100
20:18:57 MQT: stat/test/RESULT = {"Var3":"100"}
20:18:57 RUL: VAR3#STATE>%VAR1% performs "backlog event threshold=100"
20:18:57 MQT: stat/test/RESULT = {"Event":"Done"}
20:18:57 RUL: EVENT#THRESHOLD performs "backlog knxtx_val1 100;SCALE1 100,0,100,0,105;SCALE2 100,0,100,0,95;add1 2;sub2 2"
20:18:58 MQT: stat/test/RESULT = {"KnxTx_Val1":"100"}
20:18:58 MQT: stat/test/RESULT = {"Scale1":"105.000"}
20:18:58 MQT: stat/test/RESULT = {"Scale2":"95.000"}
20:18:58 MQT: stat/test/RESULT = {"Add1":"107.000"}
20:18:58 MQT: stat/test/RESULT = {"Sub2":"93.000"}
20:19:11 CMD: var3 105
20:19:11 MQT: stat/test/RESULT = {"Var3":"105"}
20:19:15 CMD: var3 106
20:19:15 MQT: stat/test/RESULT = {"Var3":"106"}
20:19:32 CMD: var3 108
20:19:32 MQT: stat/test/RESULT = {"Var3":"108"}
20:19:32 RUL: VAR3#STATE>%VAR1% performs "backlog event threshold=108"
20:19:32 MQT: stat/test/RESULT = {"Event":"Done"}
20:19:32 RUL: EVENT#THRESHOLD performs "backlog knxtx_val1 108;SCALE1 108,0,100,0,105;SCALE2 108,0,100,0,95;add1 2;sub2 2"
20:19:32 MQT: stat/test/RESULT = {"KnxTx_Val1":"108"}
20:19:32 MQT: stat/test/RESULT = {"Scale1":"113.400"}
20:19:32 MQT: stat/test/RESULT = {"Scale2":"102.600"}
20:19:32 MQT: stat/test/RESULT = {"Add1":"115.400"}
20:19:32 MQT: stat/test/RESULT = {"Sub2":"100.600"}
20:19:45 CMD: var3 100
20:19:45 MQT: stat/test/RESULT = {"Var3":"100"}
20:19:46 RUL: VAR3#STATE<%VAR2% performs "backlog event threshold=100"
20:19:46 MQT: stat/test/RESULT = {"Event":"Done"}
20:19:46 RUL: EVENT#THRESHOLD performs "backlog knxtx_val1 100;SCALE1 100,0,100,0,105;SCALE2 100,0,100,0,95;add1 2;sub2 2"
20:19:46 MQT: stat/test/RESULT = {"KnxTx_Val1":"100"}
20:19:46 MQT: stat/test/RESULT = {"Scale1":"105.000"}
20:19:46 MQT: stat/test/RESULT = {"Scale2":"95.000"}
20:19:46 MQT: stat/test/RESULT = {"Add1":"107.000"}
20:19:46 MQT: stat/test/RESULT = {"Sub2":"93.000"}
20:19:58 CMD: var3 99
20:19:58 MQT: stat/test/RESULT = {"Var3":"99"}
20:20:04 CMD: var3 80
20:20:04 MQT: stat/test/RESULT = {"Var3":"80"}
20:20:04 RUL: VAR3#STATE<%VAR2% performs "backlog event threshold=80"
20:20:04 MQT: stat/test/RESULT = {"Event":"Done"}
20:20:04 RUL: EVENT#THRESHOLD performs "backlog knxtx_val1 80;SCALE1 80,0,100,0,105;SCALE2 80,0,100,0,95;add1 2;sub2 2"
20:20:04 MQT: stat/test/RESULT = {"KnxTx_Val1":"80"}
20:20:04 MQT: stat/test/RESULT = {"Scale1":"84.000"}
20:20:04 MQT: stat/test/RESULT = {"Scale2":"76.000"}
20:20:04 MQT: stat/test/RESULT = {"Add1":"86.000"}
20:20:04 MQT: stat/test/RESULT = {"Sub2":"74.000"}
```

So, please, test those rules and report back if they worked. If so, we can add them to the docs as another example in the KNX section if you like.

Winni66, 2020-04-17:
Hello Adrian,
your proposal seems to work fine, thanks a lot. Nevertheless I have still one strange behaviour, but I don't know whether it's linked to Tasmota or to my KNX-line-setup. I get all the power information three times.
[image]
I think is linked to "Communication enhancement" ? I activated this, because it's needed sometimes for power on/off information. A loss of an information there is critical. After deactivating this, I get the value only once.
Do you have any proposal how to solve this?

ascillato, 2020-04-17:
Exactly. The "Communication enhancement" is just that. It sends 3 times the same value so as to ensure that the value reaches the receptor. This "trick" makes the wifi router to not discard the multicast packet. This is not happening in all routers, but some of them give very little priority to transmitting multicast and some messages are just discarded if the router is busy. This "enhancement" avoids that. It is kind of similar to the enhacement of Gira KNX Gateways, but simplier. As these telegrams are very small, they won't affect the performance of your KNX Network. I have tested that with a huge number of telegrams and devices and there is no issue. You can have a lot of Tasmota KNX devices communicating between them.
So, those telegrams are intended to be there.

Winni66, 2020-05-14:
Hello Adrian,
my I use this troubleshooting thread again. I have still much traffic on the bus (and I mean KNX, not IP because I have to route all this information also into KNX bus and I want to have several plugs in my house. 
Is there any opportunity, that values which are only set for more the two seconds are taken into account? I have very often short peaks in the values of those gosund plugs (currently working with new Tasmota 8.3, but it was the same in the past). 
There are almost every time two values almost immediately one after the other with a certain percentage decrease. Maybe there is an issue with calibration with powerset ? What does Powersetcal 12530 mean? In the main menu of tasmota, there are values shown (stable) from 218W to 223W, On KNX bus there are always 240W to 260W and after that immediately a downgrade to the shown round about 220W. How can I remove the calibration values to see wether it will disappear?
Thanks again for you patience and support
[image]

ascillato, 2020-05-17:
Hi, sorry for the late respond. Let me check it.

Winni66, 2020-05-25:
Hi Adrian, did you have the chance to test, whether the issue is linked to calibration? Or can you tell me how to reset calibration?

ascillato, 2020-05-27:
Hi, Sorry, couldn't make any time for test that. I want to, I just couldn't. Will do ASAP.

ascillato, 2020-05-27:
Hi, I couldn't reproduce your issue with a Pow R2.

How is your configuration in your device? Can you post a screenshot of your Tasmota KNX config page? Can you share your rules? Are you using powerdelta?

The poweset command is a calibration command. It is not related to your spikes. All values sent by KNX are the same that are being shown in the webpage (all values are after calibration, there are no raw values shown or sent)

The spikes you see are the ones measured by your device. In the webpage, the values are the same but may be you can't see them because they are just updated every few second and not instantaneously. Also, every power measurement takes 1 second or so, to be performed.

As you have different values very close to each other, I assume you are using rules or powerdelta, in that case, you need to adjust your rules or powerdelta in order to decrease the amount of messages.

ascillato, 2020-05-31:
Hi,

Any news on this?

ascillato, 2020-06-03:
What I need for having the complete picture of your setup is everything that is asked in the template when opening a new issue. If you want, you can open a new issue with all that completed.

Winni66, 2020-06-03:
Hello Adrian,
sorry for late answer, but I didn't had time to answer immediately, and after that I lost track of the issue.

First of all, I did a test with a simple light bulb (150W), there I had no issue, everything is fine. 
But with my PC at the plug it's changing really quite often, there seems to be peaks in comsumption.
The rule I currently use, is the one you proposed:

on system#boot do backlog var1 0;var2 0 endon on Energy#Power>%var1% do event threshold=%value% endon on Energy#Power<%var2% do event threshold=%value% endon on event#threshold do backlog knxtx_val1 %value%;SCALE1 %value%,0,100,0,107;SCALE2 %value%,0,100,0,93;add1 2;sub2 2 endon

So, behaviour of Tasmota seems to be correct, question remains whether there is a possibility with rules to skip such peaks (less then 1 or 2 seconds).

best regards, and thanks for this great support and the great KNX-support of Tasmota

ascillato, 2020-06-03:
> question remains whether there is a possibility with rules to skip such peaks

The easiest way is to increase the threshold beyond the value of the peaks.

A more complex approach could be calculating the average and sending that to KNX.

Another could be reading the energy value and storing it for 2 seconds. If no new value is received (meaning no spike) you send that value over KNX. If a new value is received BEFORE those 2 seconds have passed, it can start over the 2 seconds count and if nothing new is received, you can send that value.
With this last approach I see that if the Power is changing all the time, nothing is sent, so we can add a 5 seconds window of sending the last value if the 2 seconds windows is being restarted over and over.
The drawdown of this is just the 1 or 2 seconds delay from the real power (besides the 1 second delay from the measurement)

I don't know how do you want/prefer to solve this?

Winni66, 2020-06-03:
Just for information: Threshold parameters are working also for KNX, not only for MQTT?

I'm quite lucky with the solution of your proposed rule.
For my peak-problem I would like a solution, that values, which should be sent according to the rule, should be at least 2 seconds outside of the given range of the rule For sure, there will be an additional delay, but as long as it's part of the rule, it should be ok for all of us.
I personally would not setup all my devices with this behaviour, because in most cases the standards solutions fits perfectly.
So if this cannot be solved without much effort, please leave it as it is. It's nothing I cannot live with.
Unfortunately I didn't do the light bulb test before and thought of false behaviour because of "PowerSet"

kind regards

ascillato, 2020-06-03:
> Just for information: Threshold parameters are working also for KNX, not only for MQTT?

Yes. If you use powerdelta, it is going to be sent by KNX too.

> For my peak-problem I would like a solution, that values, which should be sent according to the rule, should be at least 2 seconds outside of the given range of the rule For sure, there will be an additional delay, but as long as it's part of the rule, it should be ok for all of us.

Ok, let me see. It should be a good complex example to add to the docs.

ascillato, 2020-06-11:
Sorry for the delay. Please test the following. I already tested but I would like to know if this works fine in your setup.

The proposed approach:

Reading the energy value and storing it for 2 seconds. If no new value is received (meaning no spike) you send that value over KNX. If a new value is received BEFORE those 2 seconds have passed, it can start over the 2 seconds count and if nothing new is received, you can send that value.
With this last approach I see that if the Power is changing all the time, nothing is sent, so we can add a 5 seconds window of sending the last value if the 2 seconds windows is being restarted over and over.
The drawdown of this is just the 1 or 2 seconds delay from the real power (besides the 1 second delay from the measurement)

The proposed rules:

```lua
rule1
  on system#boot do backlog var1 0;var2 0;var3 0;var4 0;var5 0 endon
  on Energy#Power>%var1% do event threshold=%value% endon
  on Energy#Power<%var2% do event threshold=%value% endon
  on event#threshold do backlog var3 %value%; ruletimer1 2;var4 0 endon
  on var4#state=%var5% do backlog ruletimer2 5;var5 1 endon
  on rules#timer=1 do backlog ruletimer2 0;event senddata endon
  on rules#timer=2 do backlog event senddata endon
  on event#senddata do backlog var5 0;knxtx_val1 %var3%;SCALE1 %var3%,0,100,0,107;SCALE2 %var3%,0,100,0,93;add1 2;sub2 2 endon
```

All together for copy & paste:

```lua
rule1 on system#boot do backlog var1 0;var2 0;var3 0;var4 0;var5 0 endon on Energy#Power>%var1% do event threshold=%value% endon on Energy#Power<%var2% do event threshold=%value% endon on event#threshold do backlog var3 %value%; ruletimer1 2;var4 0 endon on var4#state=%var5% do backlog ruletimer2 5;var5 1 endon on rules#timer=1 do backlog ruletimer2 0;event senddata endon on rules#timer=2 do backlog event senddata endon on event#senddata do backlog var5 0;knxtx_val1 %var3%;SCALE1 %var3%,0,100,0,107;SCALE2 %var3%,0,100,0,93;add1 2;sub2 2 endon
```

_Note: This rule is long, so, if you use latest Tasmota it will get compressed automatically and will fit in one rule set. But if you use an old version of Tasmota you should separate it and use 2 rule slots (ensure you enable both slots)._

ascillato, 2020-06-11:
If you want simplier rules, we can eliminate the 5 seconds window and put `teleperiod` to 10 seconds for that.

Winni66, 2020-06-11:
Hello Adrian,
Even if I don't understand all of your rule ;-) I tried immediately.
[image]
But I think I expected another behaviour (please ignore mA information).
There are many entries, where the difference is lower then 14% (ca. 30W) 
But there is now much less traffic on the bus than before. I will try to rise seconds (for this special gosung plug for my PC). 
This is just for getting feedback to your efforts, there is no need to go in more details if you are not interested on this.

ascillato, 2020-06-11:
A PC has a variable power consumption, so, it is expected to see that. You can increase the thresholds if you want.

> only the values which are (almost) identical and coming in about 10 seconds one after the other are confusing me

If the power has several spikes one after another, and it don't stabilize inside the thresholds, the rule will post regularly the actual value like a teleperiod.

> Did some changes on the parameters and I'm really satisfied

Cool !!!! :+1:

closed 2020-06-11 by ascillato