ascillato/Tasmota_KNX #151: New proposal of logo for Sonoff-Tasmota_KNX

orlant, 2018-07-05:
HELLO I AM A GRAPHIC DESIGNER AND I LIKE TO MAKE CONTRIBUTIONS IN PROJECTS LIKE HIS. If you allow me, I can present you a LOGO-MARK for your project Sonoff-Tasmota_KNX

ascillato, 2018-07-05:
Wow, great! 

If you want, I can add a logo made by you to this project. I would love that. 

But remember that Sonoff-Tasmota_KNX is a branch of Sonoff-Tasmota 

At this moment, Sonoff-Tasmota and Sonoff-Tasmota_KNX are exactly the same (just a couple of custom options explained in releases page) but essentially the same. I'm the owner of this branch, but also one of the collaborators of the original branch of Sonoff-Tasmota. 

If you want, we can present that in the original branch of Sonoff-Tasmota. I think all the people there would love your idea to add a logo.

What do you think? And what logo do you have in mind?

ascillato, 2018-07-05:
Your Sonoff-Tasmota logo or similar also could be used in other related projects:

TasmoAdmin
Sonota

orlant, 2018-07-05:
hello Mr. Ascillato sorry I did not return before, I was just feeding all the necessary information about Sonoff-Tasmota and its alternative Firmware version KNX ... I noticed that the main idea Sonoff-Tasmota has no corporate image either ... but the accessories Sonoff yes, then I will be careful not to interfere between one and the other ...

For now I have some things in mind, but I have to order them with your help to get the best result. We can start designing based on the update of Sonoff-Tasmota by Theo Arends and from there, we will have the much easier branching of the Logo of Sonoff-Tasmota_KNX. I'll be back soon with some ideas. the logo will be reactive, I feel inspired by the combination of letters and symbols that represent firmware, software and technology ... it will be a pleasure to work for you.

graphic designer orlando velasquez
orlant90@gmail.com

ascillato, 2018-07-05:
Remember that the original Sonoff-Tasmota is KNX now.

Sonoff-Tasmota = Sonoff-Tasmota_KNX

So, a logo for Sonoff-Tasmota should be the same than for Sonoff-Tasmota_KNX because both are the same thing. 

This repository is the one I use for development of new KNX features and test but then, all that work is included in the original Sonoff-Tasmota

I'm looking forward to see your logo :+1:

orlant, 2018-07-05:
yes thank you very much I already have it in mind. I will soon return a design that catches the attention of all

orlant, 2018-07-05:
For me, it is a pleasure to introduce you to the idea arising from the union of the initials of Sonoff Tasmota.

S + T + network design. ... this logo is very easy to recognize. even if we remove the letters you can use it with an ideal LogoMark for that project

[image]
[image]

application of the logo in the firmware
[image]
[image]

I hope you like my idea, any modification of colors and font you want I am ordering ... there are both versions.

ascillato, 2018-07-05:
Love it! :+1:

Can you post this on the main branch of Sonoff-Tasmota so Theo (the owner of Tasmota) can take a look at this? 

https://github.com/arendst/Sonoff-Tasmota

ascillato, 2018-07-05:
Theo is the owner and he has several years invested in this project, so I'm sure that he will have more ideas to share with you.

orlant, 2018-07-06:
yes perfect ... but your personal opinion that you think about the design? It is at the height? Do you think it could be the logo of your KNX version?

ascillato, 2018-07-06:
Hi,

This is a fork (not the original), I was trying to explain. I use this repository for testing. The original Sonoff-Tasmota software is at https://github.com/arendst/Sonoff-Tasmota

I'm not the owner of Tasmota. Theo is the owner. He is the one that should decide everything related to Tasmota.

Your logo is nice but I'm not the one who decide to be or not to be included. Theo is.

The original Tasmota is also KNX. 

This repository is the same as Sonof-Tasmota

Let's see what Theo has to say about a logo.

_Please, change the title of the issue that you made at https://github.com/arendst/Sonoff-Tasmota
- you put it in spanish - need to be in english. Also, PLEASE, clarify that it is your logo, your work. It is not my logo request as you put there. I do not want to steal the credit for you. The logo is all your creation and your creativity. I have nothing to do with that - You can edit your comment there.._

Closing the issue. Let's continue in the original main branch.

closed 2018-07-06 by ascillato

orlant, 2018-07-06:
Thanks for your corrections I had a little time on github and there are some functions and terms that I do not master much but I have already made all the corrections that you have asked me to verify with pleasure. thank you very much for your kindness