ascillato/Tasmota_KNX #230: Rule to send KNX Telegramm with BH1750 Sensor Data

lost-hope, 2018-09-22:
Hello
I have a little Problem with the BH1750 Sensor.
It is set up and works in tasmota
[image]

Now i want to send the Sensor Data through KNX. I did not get even a rule to work to save the sensordata to a Variable, when the value is different from the variable.
Here is my attempt of the rule:
on BH1750#Illuminace!=%var1% do var1 %value% endon

I am not quite shure if the != works in tasmota rules, but also a > did not work
Can you please Help?
When this is not working with KNX it is also possible to sent it to MQTT konstantly and i will then adapt it to KNX on change.

The Hardware setup is a bit strange, but i don't want to tinker around with 230V.

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [x] Hardware used : ESP8266 running Tasmota
                                BH1750 

- [x] Provide the output of command ``status 0`` :
```
10:54:25 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:54:25 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:29:23","StartupUTC":"2018-09-22T09:25:02","Sleep":0,"BootCount":25,"SaveCount":91,"SaveAddress":"F9000"}}
10:54:25 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018.08.31 20:36:46","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
10:54:25 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXX",""],"TelePeriod":300,"SetOption":["02008001","55818000","00000000"]}}
10:54:25 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":0,"Features":["00000407","1F90A984","0C0003B0","23021796","00000000"]}}
10:54:25 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-7654","IPAddress":"XXX.XXX.XXX.XXX","Gateway":"XXX.XXX.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer":"XXX.XXX.XXX.XXX","Mac":"XXXXXXXXX","Webserver":2,"WifiConfig":5}}
10:54:25 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 22 09:54:25 2018","Local":"Sat Sep 22 10:54:25 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:36","Sunset":"18:49"}}
10:54:25 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-09-22T10:54:25","DHT11":{"Temperature":21.9,"Humidity":55.0},"BH1750":{"Illuminance":462},"TempUnit":"C"}}
10:54:25 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-09-22T10:54:25","Uptime":"0T00:29:23","Vcc":2.980,"POWER":"OFF","Wifi":{"AP":1,"SSId":XXXXXXXX","RSSI":50,"APMac":"XXXXXX"}}
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-09-22:
Hi,

Sorry, the operand != is not used for rules triggers, so that rule will not work.

* If you want to send your sensor values by KNX **every teleperiod time** to the Group Address defined in KNX_TX1, you can use the following rule:

```
rule1 1
rule1 on tele-BH1750#Illuminace do knxtx_val1 %value% endon
```

* If you want to send your sensor values by KNX only **when it changes in a delta of 10 lx** to the Group Address defined in KNX_TX1, you can use the following rule:

```
rule1 1
rule1 on system#boot do backlog var1 0; var2 0 endon on BH1750#Illuminace>%var1% do backlog var1 %value%; knxtx_val1 %value%; var2 %value%; add1 2; sub2 2 endon on BH1750#Illuminace<%var2% do backlog var2 %value%; knxtx_val1 %value%; var1 %value%; add1 2; sub2 2 endon
```

lost-hope, 2018-09-23:
Hello
it worked, but it is spelled Illuminance so there is a missing n

Greetings 
Sören

closed 2018-09-23 by lost-hope

ascillato, 2018-09-23:
Thank you. Typo corrected. :+1: