ascillato/Tasmota_KNX #318: Sonoff SC with addional snesor (DHT22)

johannesbonn, 2018-11-26:
Hello,

I have the following problem. I am using a Sonoff SC with an addional sensor (type DHT 22 temperature and humidity). But I can not chose this two values for the knx configuration. Only the internal sensor is possible to chose.

Is it possible that you add also an internal sensor and his channels for the knx configuration.

Thank you very much!

ascillato, 2018-11-26:
Hi,

By now, there is no direct way to configure that from the KNX menu.

But to transmit now that sensor over KNX, you can use **rules** and the `KnxTx_val1` command. There is an example of this in the Tasmota wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/KNX-Features#5-using-rules

Similar issues:

https://github.com/ascillato/Sonoff-Tasmota_KNX/issues/140
https://github.com/ascillato/Sonoff-Tasmota_KNX/issues/230

Hope this helps

closed 2018-11-26 by ascillato

ascillato, 2018-11-27:
If you need further assistance, please, do not hesistate and find us in the Official Tasmota Support Chat at https://discord.gg/Ks2Kzd4