ascillato/Tasmota_KNX #348: Temperature is send every 3 Seconds (with rule to send it per KNX)

johannesbonn, 2019-01-07:
Hello,

I have the following problem with my wemo d1 mini (Tasmota 6.3.1).
I have a DHT22 sensor connected and with a rule I send the values to my KNX Gateway.The Problem is that the wemo sends the value for temperature every 3 seconds, but not for humidity.

`19:39:00 RUL: AM2301-02#TEMPERATURE performs "knxtx_val1 6.1"
19:39:00 KNX: KNX TX 1 = 6.10 sent to 0.0.4
19:39:00 RSL: RESULT = {"KnxTx_Val1":"6.10"}
19:39:04 RUL: AM2301-02#TEMPERATURE performs "knxtx_val1 6.1"
19:39:04 KNX: KNX TX 1 = 6.10 sent to 0.0.4
19:39:04 RSL: RESULT = {"KnxTx_Val1":"6.10"}
19:39:07 RUL: AM2301-02#TEMPERATURE performs "knxtx_val1 6.1"
19:39:07 KNX: KNX TX 1 = 6.10 sent to 0.0.4
19:39:07 RSL: RESULT = {"KnxTx_Val1":"6.10"}`

What is wrong?

Thank you for your help.

ascillato, 2019-01-07:
Please, if you are using rules, share your rules

johannesbonn, 2019-01-08:
my rule
` {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":459,"Rules":"on AM2301-02#Temperature do knxtx_val1 %value% endon"}`

teleperiod is set to 60 seconds. I Need the rule because I have 2 sensors on my wemo.

thank you!

ascillato, 2019-01-09:
Ok. That's it. You have only one rule and just for temperature. Just add another rule for humidity.
Remember that you need to set a different group address to send humidity. You can not send both to the same.

ascillato, 2019-01-09:
I kept thinking about you said you have 2 sensors in your wemos. Tasmota supports only one DHT. If you have 2 DHT only one will work. If you have 2 DS18S20, only the first will be in the Tasmota KNX Menu. For the second you need to use rules.

Anyway, if you need further assistant, please find us in the Tasmota Chat. It is better to discuss troubleshooting there. It is more dynamic.

Another thing. This repo is a fork from the original Tasmota. At this moment it is exactly the same than the original Sonoff-Tasmota. If you post an issue here, I will be the only one seeing it. If you want faster response, may be it is better to address any Tasmota issue in the chat or in the original repository. I have no problem, it is just an advise for faster response because in the chat and in the original repository, there are a lot of people helping. 

See you.

johannesbonn, 2019-01-09:
For sure I have a second rule for humidity, but only temperature is send every 3 seconds to the bus. And for sure I have different addresses.

Thank you!

ascillato, 2019-01-09:
Ok, so. Please, share all your rules to check them

johannesbonn, 2019-01-09:
I have only two rules. The second one (humidity):
`{"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":457,"Rules":"on tele-AM2301-02#Humidity do knxtx_val2 %value% endon"}`
 
The first sensor is recognized as Temperature and Humidity sensor for the KNX part.

Thank you!

johannesbonn, 2019-01-09:
and I have one knx adress for each sensor item (4 adresses)

ascillato, 2019-01-09:
Ok. Let me do some tests. You have one "tele-" event that is been triggered but two calls to that event.

When it is a teleperiod time, it triggers the tele- message for all sensors but you have 2 items for the same sensor. Let me test and I will come back to you. Thanks

johannesbonn, 2019-01-10:
OK, thank you for your effort!

johannesbonn, 2019-01-17:
Hi Adrian,
any news to this topic?

ascillato, 2019-01-17:
Sorry, I didn't have time. Don't worry, I will look into it. ;)

ascillato, 2019-01-28:
Sorry, these days I'm busy. Don't worry, I will look into it. ;)

johannesbonn, 2019-01-31:
Ok, thanks for remind me. I reflashed the Wemo and with the same firmware 6.3 and rules no more problems. But the reason I don’t know.
 Thank you for you effort.

ascillato, 2019-02-02:
Sorry I couldn’t test it, I was super busy with urgent stuff. May be was just a typo somewhere. Anyway, if you have this issue again, let's reopen this issue. Thanks.

closed 2019-02-02 by ascillato