ascillato/Tasmota_KNX #563: Tasmota KNX missing functions.

WoutervanZijl, 2020-11-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Currently i have a few device with Tasmota KNX but i canot find any option to choose like relay or button or anything,
The only option i have is to select a output in the configuration menu of knx.
My modules are Sonoff 4 ch pro and 3 sonoff basics all with tasmota knx 9.1.
I have searched online but only see picture on the tasmota knx wiki where there is a button selected or humidity and other option. Am i doing something wrong or is there more info hidden or coming.

What i would like if possible is to program my sonoff 4ch pro so that my rf buttons are seperate from the sonoff device to start scene or indeed switch a relay so that its more programmeble to amazing idees:)

I realy like knx and if I could help in any way or test ( not a programmer but a enthousiast for sure)

Hope this the correct way of communicating because i dit tryed the chat but no luck there.
 
 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ___sonoff 4ch pro and sonoff basic__
- [ ] Tasmota binary firmware version number used: ___9.1__tasmota knx
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```this is where i dont know anything about and schould need to learn more about and where i can learn that.
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-11-21:
> ... i canot find any option to choose like relay or button or anything ...

The available options for KNX Web Menu, are only the ones defined at https://tasmota.github.io/docs/KNX/#implemented-features

To be able to select any of those, you must have them defined under MODULE CONFIGURATION. If your device have a relay and you can operate it using the main page of Tasmota, you will have it as OUTPUT in the KNX web menu.

> What i would like if possible is to program my sonoff 4ch pro so that my rf buttons are seperate from the sonoff device to start scene or indeed switch a relay

To use the RF buttons for scenes, you need to use **_rules_**. You can use the `KnxTx_scene` command in those rules to send a KNX scene.

ascillato, 2020-11-24:
If yoy have further questions, please just ask.

closed 2020-11-24 by ascillato

WoutervanZijl, 2020-11-25:
Thank you very much for you’re answer. I got it working now🥳🥳but the knx version that is different when i start this. 9.1.0 does not have buttons. Then is clicked another link to a other binaries. Tasmota knx 9.1.0.2 does have buttons. But 5 minutes ago after more googling i found diferent page with knx with language and version 9.0.0.2

Keep using this page for the best experience? Where version 9.0.0.2 is found?

Also I tried to add a device with ets5 demo.
I could not get it working. How to ad a dummy (i have added a gira dummy, but then I got stuck🙈 I think I need to google more about ets5 anyway. 

I you have advice to get it running with basics in knx5, I would love to hear. 

Very nice to use a button and more lichts go on🥳

Great firmware.

ascillato, 2020-11-25:
Tasmota main repository is https://github.com/arendst/Tasmota

The latest Tasmota version is at http://ota.tasmota.com/tasmota

Also you can set into your Tasmota (from upgrade firmware in the webmenu) the latest KNX Tasmota version with http://ota.tasmota.com/tasmota/tasmota-knx.bin.gz so you can just do one click and Tasmota will take care automatically of all the update process (download, free space in flash, flash the new file, etc.)

This repository is a fork that I use to develop new KNX functions for Tasmota. For testing I also share the precompiled bins in all supported languages at https://github.com/ascillato/Tasmota_KNX/releases/latest but they are not the latest. Tasmota is being updated nearly everyday. The latest Tasmota with all the fixes (including KNX fixes) is at the official http://ota.tasmota.com/tasmota/tasmota-knx.bin.gz

For adding a dummy switch in ETS, please search in YouTube. There are plenty of videos showing exactly how to add it.

WoutervanZijl, 2020-11-25:
Thank you for responding.
Do I need a Knx ip interface or router to get a connection?
If I google I only find stuf about interfaces or routers.

ascillato, 2020-11-25:
If you have a KNX network on TP(twisted pair comms), RF(radio frequency comms) or PL(power line comms), you need an interface to connect your knx network to your LAN network.

Tasmota uses the KNX IP protocol, so it needs just wifi to communicate to other Tasmotas, but if you want to interact to your KNX network you need to connect it to your wifi/lan network. Also, the KNX IP interface must support KNX IP Multicast to use Tasmota. As far as I know, nowadays all LAN-KNX interfaces supports KNX IP Multicast. KNX Tunneling is not supported in Tasmota.

WoutervanZijl, 2020-11-25:
So if I understand it correctly. To configure tasmota with ETS5 I need a knx interface? Or can I connect to Tasmota different? I have no any other knx devices. Only Tasmota. 

I my head I thought i would just connect to the same wifi network as Tasmota knx and Add a device and configure it.

ascillato, 2020-11-25:
To configure Tasmota you only need to access Tasmota webUI. No need for anything else.

WoutervanZijl, 2020-11-25:
Yes i have seen that. But If I would like to play with it more and ets5 is it possible without interface or router to connect?

ascillato, 2020-11-25:
Also, if you have a Home Automation software like for example HOME ASSISTANT, you can make your devices to communicate to that using KNX IP. Also you can do the same using MQTT.

ascillato, 2020-11-25:
> But If I would like to play with it more and ets5 is it possible without interface or router to connect?

Tasmota can't be configured from ETS. Tasmota KNX must be configured using its own web menu.

WoutervanZijl, 2020-11-25:
Yes When i learn more i will use home assistand indeed. I like knx more and its been here for long time. Maybe expanding with hager and knx rf

ascillato, 2020-11-25:
ETS is for configuring KNX hardware. Depending on your KNX Network type, is the adapter you need. For example, if you have KNX IP devices, you need just a PC connected to your wifi router. If you have a KNX-TP, you need the serial to TP adapter or any other adapter to communicate to KNX TP and config it.

WoutervanZijl, 2020-11-25:
> > But If I would like to play with it more and ets5 is it possible without interface or router to connect?
> 
> Tasmota can't be configured from ETS. Tasmota KNX must be configured using its own web menu.

Ah oke nice to know. What is possible with ets if i make a dummy device. Only for connecting commands to an from a different knx line?

ascillato, 2020-11-25:
For KNX RF you need indeed an interface but that is not related to Tasmota. Tasmota offers a support for some functionalities over KNX IP in order to be able to accept and sends commands to other Tasmotas and to KNX devices, but is not a KNX hardware, so it isn't configurable under ETS.

ascillato, 2020-11-25:
> What is possible with ets if i make a dummy device.

Nothing. Tasmota can't be configured from ETS. Only from its own web menu.

WoutervanZijl, 2020-11-25:
Ah oke nice. I thought maybe to get a cheaper knx system in my house 😁

One question about reading a device. If i program a button to a relay en then the output to another tasmota and then back to again a different relay. If i press realy fast  twice then they are not all on. the first i press is out again. Is it possible to read the status?

ascillato, 2020-11-25:
The dummy device in ETS is only useful when you are programming an entire KNX network. For example, If you are telling in ETS that a KNX switch must send an ON command to the Group address 2.2.2 and that GA is not related to anything (only your Tasmota device), ETS will tell you that the GA 2.2.2 is not associated to anything. So, for eliminating that error, you must tell ETS that GA 2.2.2 is a dummy device.

WoutervanZijl, 2020-11-25:
> The dummy device in ETS is only useful when you are programming an entire KNX network. For example, If you are telling in ETS that a KNX switch must send an ON command to the Group address 2.2.2 and [...]

Yes clear thank you!

ascillato, 2020-11-25:
> I thought maybe to get a cheaper knx system in my house

You can, with Tasmota you can have a mixed device type network. Using KNX or MQTT.

Choosing  KNX or any other Smart Home system is just about how are the properties you want. How reliable it is, its support, its expandability, if it supports diy devices, and price.

ascillato, 2020-11-25:
>One question about reading a device. If i program a button to a relay en then the output to another tasmota and then back to again a different relay. If i press realy fast twice then they are not all on. the first i press is out again. Is it possible to read the status?

Hehehe, you need to test that. Follow the example on the Tasmota docs and you will see. It is pretty reliable the comms.

And yes, you can read the actual status of a relay.

ascillato, 2020-11-25:
If you don't have any KNX device, you can evaluate if it is the right choice for you or not.

If you just wanted to play with ETS and a KNX device, Tasmota is not a solution to that for now. If you want that, I can search in my notes that there is another library that supports ETS programming. It isn't as complete as Tasmota and you will need to code and compile to use it.

WoutervanZijl, 2020-11-25:
> Hehehe, you need to test that. Follow the example on the Tasmota docs and you will see. It is pretty reliable the comms.
>
> And yes, you can read the actual status of a relay.

I did and it worked great😁 But I noticed that. Because when I use it in the staircase and the button is pushed realy fast its the other way around. I wil test more and find out👌

WoutervanZijl, 2020-11-25:
> If you don't have any KNX device, you can evaluate if it is the right choice for you or not.
> 
> If you just wanted to play with ETS and a KNX device, Tasmota is not a solution to that for now. If you want that, I can search in my notes that there is another library that supports ETS programming. It isn't as complete as Tasmota and you will need to code and compile to use it. Tell me if you want that. Is in github.

Oh I am am not experienced in what to do with code🙈

WoutervanZijl, 2020-11-27:
> The available options for KNX Web Menu, are only the ones defined at https://tasmota.github.io/docs/KNX/#implemented-features
>
> To be able to select any of those, you must have them defined under [...]

Hey ascillato, is it possible to use the rf buttons separate from the sonoff switxh itself? Normaly you link a button pres to the sonoff switch 1 and when switching it goes on. Can this be seperat
So that when button pressed the switch does not toggle
But does
Send a message to knx.

How exactly can I add rf to a tasmota scene

ascillato, 2020-11-27:
Please, tell me which device is an also please, set weblog to 4 and paste here the output of the console when you receive the signal from the rf. Thanks.

WoutervanZijl, 2020-11-28:
> Please, set weblog to 4 and paste here the output of the console when you receive the signal from the rf. Thanks.

Yes I did what you asked.

Saved to flash at F4, Count 414, Bytes 4096
09:10:05 WIF: Checking connection...
09:10:18 APP: Button1 level 1-0
09:10:18 KNX: Button 1 = 1 sent to 1.1.10
09:10:18 SRC: Button
09:10:18 RSL: RESULT = {"POWER1":"ON"}
09:10:18 RSL: POWER1 = ON
09:10:19 CFG: Saved to flash at FB, Count 415, Bytes 4096
09:10:19 APP: Button1 level 0-1
09:10:25 WIF: Checking connection...
09:10:35 SRC: WebGui from 192.168.1.5
09:10:35 RSL: RESULT = {"POWER1":"OFF"}
09:10:35 RSL: POWER1 = OFF
09:10:36 CFG: Saved to flash at FA, Count 416, Bytes 4096
09:10:37 HTP: Configuration
09:10:41 HTP: Main Menu
09:10:42 HTP: Console
09:10:45 WIF: Checking connection...
09:10:54 APP: Button1 level 1-0
09:10:54 KNX: Button 1 = 1 sent to 1.1.10
09:10:54 SRC: Button
09:10:54 RSL: RESULT = {"POWER1":"ON"}
09:10:54 RSL: POWER1 = ON
09:10:54 APP: Button1 level 0-1
09:10:55 CFG: Saved to flash at F9, Count 417, Bytes 4096
09:11:05 WIF: Checking connection...
09:11:08 APP: Button1 level 1-0
09:11:08 KNX: Button 1 = 1 sent to 1.1.10
09:11:08 SRC: Button
09:11:08 RSL: RESULT = {"POWER1":"OFF"}
09:11:08 RSL: POWER1 = OFF
09:11:08 APP: Button1 level 0-1
09:11:09 CFG: Saved to flash at F8, Count 418, Bytes 4096
09:11:22 HTP: Main Menu
09:11:25 WIF: Checking connection...
09:11:27 APP: Button3 level 1-0
09:11:27 SRC: Button
09:11:27 RSL: RESULT = {"POWER3":"ON"}
09:11:27 RSL: POWER3 = ON
09:11:27 APP: Button3 level 0-1
09:11:28 CFG: Saved to flash at F7, Count 419, Bytes 4096
09:11:28 APP: Button4 level 1-0
09:11:28 SRC: Button
09:11:28 RSL: RESULT = {"POWER4":"ON"}
09:11:28 RSL: POWER4 = ON
09:11:29 CFG: Saved to flash at F6, Count 420, Bytes 4096
09:11:29 APP: Button4 level 0-1
09:11:32 HTP: Console
09:11:45 WIF: Checking connection...
09:12:05 WIF: Checking connection...
09:12:25 WIF: Checking connection...
09:12:45 WIF: Checking connection...
09:13:05 WIF: Checking connection...

I know i can send a command from the rf device to the sonof switch and it turns the switch on. Can I also make this seperate? 
So that 1 press button one and nothing happen but it switches a on signal to knx groep 1.1.1 for instance?

Same af with a scene. So that when its pressed button one the switchlink is off.

Cheers

ascillato, 2020-11-29:
> Can I also make this seperate? So that 1 press button one and nothing happen but it switches a on signal to knx groep 1.1.1 for instance?

In Tasmota web menu:

* go to CONFIGURATION, then CONFIGURE TEMPLATE
* select **Sonoff 4 ch pro** as *BASED ON*.
* change button1 to switch1
* go to CONFIGURE OTHER 
* select activate template.
(Now the behavior is the same as before, but your button1 can be controlled easily with rules)
* go to the console and type:
```lua
Switchmode1 0
Switchtopic 0
Rule1 1
Rule1 on switch1#state do knxtx_cmnd1 1 endon
```
* go to the KNX web menu and set KNX TX Slot1 as 1/1/1

Now every time you press button1 or the RF button1, your device won't change the state of relay1 but will send a ON to 1/1/1

(The RF button1 and the physical button1, are the same input in your device - they can not be separated as they are connected in the hardware to the same GPIO of the ESP8266)

WoutervanZijl, 2020-11-30:
> In Tasmota web menu:
>
> * go to CONFIGURATION, then CONFIGURE TEMPLATE
> * select **Sonoff 4 ch pro** as _BASED ON_.
> * change button1 to switch1
> * go to CONFIGURE OTHER
> * select activate template. [...]

When i put in switchtopic 0 
command unknown.

And for switch 2 i put in the same but then change rule1 on switch 2?

About the physical connection with rf and the button I understand. This is what i wanted thank you.

Is there progres with a dimmer with knx maybe?🙏🏻

ascillato, 2020-11-30:
> When i put in switchtopic 0, command unknown.

May there was a typo when introducing the command? It is `SwitchTopic`
Anyway, don't worry, the zero is the default, but must be zero in order to let the rules uses the switches. 

> And for switch 2 i put in the same but then change rule1 on switch 2?

Yes, exactly. 

> Is there progres with a dimmer with knx maybe?

Yes, but I do that only on my free time, that is very limited right now. I need to pay the bills ;)
So, please, be patient. Thanks.

ascillato, 2020-11-30:
You can check the progress on the dimming and RGB functions on the feature request https://github.com/arendst/Tasmota/issues/7438

WoutervanZijl, 2020-12-01:
> May there was a typo when introducing the command? It is `SwitchTopic`
> Anyway, don't worry, the zero is the default, but must be zero in order to let the rules uses the switches.
>
>
> Yes, exactly. [...]

Nice I got it to work yes💪
I feel I got in late with tasmota and stil finding how it work with rules and everything.

Nice to hear there that its in the making knx rgb!  Does that also implies for instance the schelly dimmer with knx. Just different programming

ascillato, 2020-12-01:
> Does that also implies for instance the schelly dimmer with knx.

Yes

WoutervanZijl, 2020-12-01:
Very nice Adrian👌

WoutervanZijl, 2020-12-15:
Hello Adrian, I borrowed from school alot of knx devices and also an ip interface.

Ik have a question about what i need to connect 20 tasmota knx devices to knx.
Do I require a knx router? Or is that something different for other uses?
Can i simple connect the ip interface to a wifi router en connect all de device to that wifi router?

Thanks

ascillato, 2020-12-15:
> Can i simple connect the ip interface to a wifi router en connect all de device to that wifi router?

Exactly.

WoutervanZijl, 2021-01-07:
Hi i have bought myself a sonof rf bridge. Now was I wondering how can i use more then 5 channels. I can make rules but there are only 5 slots for this. Can I expand that?

ascillato, 2021-01-07:
As you said, you can using rules.

WoutervanZijl, 2021-01-07:
Mm oke. Could you provide me a start line😁
So you mean I can make rx6 wit rules then?

ascillato, 2021-01-07:
Yes, you can adapt the RF Bridge example of the docs to send your desired commands by KNX.

https://tasmota.github.io/docs/Rules/#rf-repeater-ir-repeater

WoutervanZijl, 2021-01-09:
Thanks for all the responses.
I find it very Difficult to understand the the programming in the console. I got it working to send a knx signal from a button of the remote but it does the same on every button.
With some googling i found a diferent rule line but i can not get it to work.

18:25:35.444 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-09T18:25:35","RfReceived":{"Sync":8730,"Low":290,"High":870,"Data":"9CD468","RfKey":"None"}}
18:25:57.720 CMD: Rule1 on RfReceived#Data=94D568 do KnxTx_Cmnd1 0 endon
18:25:57.726 RUL: Stored uncompressed, would compress from 48 to 41 (-15%)
18:25:57.731 RSL: stat/tasmota_00ED6A/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":48,"Free":463,"Rules":"on RfReceived#Data=94D568 do KnxTx_Cmnd1 0 endon"}
18:26:00.583 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-09T18:26:00","RfReceived":{"Sync":8710,"Low":300,"High":870,"Data":"9CD568","RfKey":"None"}}
18:26:08.950 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-09T18:26:08","RfReceived":{"Sync":8710,"Low":300,"High":870,"Data":"9CD568","RfKey":"None"}}
18:26:13.156 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-09T18:26:13","RfReceived":{"Sync":8710,"Low":300,"High":870,"Data":"9CD568","RfKey":"None"}}

rule1 on RfReceived#Data=94D568 do KnxTx_Cmnd1 0 endon

I also dont understand how to make the rule where I can make my own rule to send knx 1/1/1

I got 5 tx slots but a 10 button remote.
Got KnxTx_Cmnd1 working but now without this for knx slot 6,7,8,9,10

Thx for the help

ascillato, 2021-01-09:
Yes, you are right. There are only 5 slots for storing KNX Group Addresses.

By now, you have reach the maximum amount of slots. This was a limitation by design due to how Tasmota was using the eeprom for settings. As it is now, this can not be increased without corrupting the settings area.

I'm working on a refactoring of the whole KNX driver and libraries to use the new way of storing settings that Tasmota has now, in order to allow more slots and add more versatility on the KNX message types. Until then, you can only use 5 slots. Sorry.

WoutervanZijl, 2021-01-10:
Don’t be sorry for that😁 the fact we can use knx for this is awesome. Ordered another sonof bridge for a solution for now.

But I have not figured out how to get the remote working.

rule1 on RfReceived do KnxTx_Cmnd1 0 endon

If i use this rule then all the 5 buttons do the same. How can I seperate this?

this rule is also not working. rule1 on RfReceived#Data=94D568 do KnxTx_Cmnd1 0 endon

what am i missing?

WoutervanZijl, 2021-01-20:
Oke I got it working to revieve my remote command from 1 button to send a knx cmd.
But when i type in the cmd for the second button it works but the my first rule disappears.🙈

RFRECEIVED#DATA=9CD569tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-20T17:16:18","RfReceived":{"Sync":8730,"Low":300,"High":870,"Data":"9CD568","RfKey":1}}
17:16:18.225 RUL: RFRECEIVED#DATA=9CD568 performs "KnxTx_Cmnd1 1"
17:16:18.231 KNX: KNX TX 1 = 1 sent to 2.1.1
17:16:18.235 RSL: stat/tasmota_00ED6A/RESULT = {"KnxTx_Cmnd1":"1"}
17:16:45.219 CMD: rule1 on RFRECEIVED#DATA=9CD564 do KnxTx_Cmnd2 2 endon
17:16:45.226 RUL: Stored uncompressed, would compress from 48 to 44 (-8%)
17:16:45.230 RSL: stat/tasmota_00ED6A/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":48,"Free":463,"Rules":"on RFRECEIVED#DATA=9CD564 do KnxTx_Cmnd2 2 endon"}
17:16:47.380 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-20T17:16:47","RfReceived":{"Sync":8720,"Low":300,"High":870,"Data":"9CD564","RfKey":2}}
17:16:47.394 RUL: RFRECEIVED#DATA=9CD564 performs "KnxTx_Cmnd2 2"
17:16:47.400 KNX: KNX TX 2 = 1 sent to 2.1.2
17:16:47.408 RSL: stat/tasmota_00ED6A/RESULT = {"KnxTx_Cmnd2":"2"}
17:16:50.076 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-20T17:16:50","RfReceived":{"Sync":8710,"Low":300,"High":870,"Data":"9CD568","RfKey":1}}
17:16:52.139 RSL: tele/tasmota_00ED6A/RESULT = {"Time":"2021-01-20T17:16:52","RfReceived":{"Sync":8710,"Low":300,"High":870,"Data":"9CD568","RfKey":1}}

What am i doing wrong?
Also i cant find any info why after knxtx_cmd there is a 1?

ascillato, 2021-01-20:
-----------
> ...but the my first rule disappears...
-----------

From the docs at https://tasmota.github.io/docs/Commands/#rules

[image]

So, you can save you rules all together as:
```lua
rule1 1
rule1 on RFRECEIVED#DATA=9CD568 do KnxTx_Cmnd1 1 endon on RFRECEIVED#DATA=9CD564 do KnxTx_Cmnd2 1 endon
```
or save one rule and append a new one as:
```lua
rule1 1
rule1 on RFRECEIVED#DATA=9CD568 do KnxTx_Cmnd1 1 endon
rule1 +on RFRECEIVED#DATA=9CD564 do KnxTx_Cmnd2 1 endon
```

You have 3 Sets for rules (RULE1, RULE2, RULE3). Each Set **_can have a lot of rules_** until there is no more free space on that rule set.
Using `RULE1 ON ....` will OVERWRITE everything you have in rule set 1. 
Using `RULE1 +ON ....` will ADD what you want to the rule set 1. 

-----------
> Also i cant find any info why after knxtx_cmd there is a 1?
-----------

From the docs at https://tasmota.github.io/docs/Commands/#knx

The command structure is

_knxtx_cmd_**X** **command**

* **X** is the KNX slot where you have set the GroupAddress to send this command. That KNX Slot must be completed by you in the Tasmota KNX web menu with the KNX address you want to send the command.
* **command** is the command you want to send. 0 for Turn it OFF and 1 for Turn it ON.

**_For example._**
If, you want to send the commands TURN ON / TURN OFF to the KNX Address 2.2.3 and 2.2.4

1- Config the addresses on the KNX Slots

[image]

2- Set the rules in the console

* code 9CD568 to turn on 2.2.3
* code 9CD567 to turn off 2.2.3
* code 9CD564 to turn on 2.2.4
* code 9CD565 to turn off 2.2.4

```lua
rule1 1
rule1 on RFRECEIVED#DATA=9CD568 do KnxTx_Cmnd1 1 endon on RFRECEIVED#DATA=9CD567 do KnxTx_Cmnd1 0 endon on RFRECEIVED#DATA=9CD564 do KnxTx_Cmnd2 1 endon on RFRECEIVED#DATA=9CD565 do KnxTx_Cmnd2 0 endon
```

3-Enjoy :)

WoutervanZijl, 2021-01-22:
Thank yo uso much!!!!!!! I got it working 😊😊😊

[image]<https://user-images.githubusercontent.com/35405447/105217500-b0f0de00-5b32-11eb-90e4-c8e13482c860.png>

[image]<https://user-images.githubusercontent.com/35405447/105219628-6fadfd80-5b35-11eb-8aaf-283eb4d02aae.png>

WoutervanZijl, 2021-01-29:
I am seeing some issue with changing knx commands. For testing purposes i had send knx code 2/1/2 to the one light so that my girlfriend was able to use it with the remote.
But after removing al the knx commands the licht stil respond to knx 2/1/2. Am I doing something wrong? Does it needs full reset first somewhere.
I had another light doing the same but i did a full reset and put new knx commands in the slot for recieving and it worked again. 
[image]

WoutervanZijl, 2021-01-29:
[image]
[image]

WoutervanZijl, 2021-01-29:
[image]

ascillato, 2021-01-29:
Sorry, I don't understand your issue. Please elaborate.

WoutervanZijl, 2021-01-29:
Wel I wil try.

I got 1 ligt up en running with tasmota.
For testing I have put group adress 2/1/2 to listen to. But now I wanted to make it permanent en removed 2/1/2 from the kist to toggle and added 3 different knx group adress to listen to but for some reason it is stil listening to 2/1/2 even its not in the list anymore. Somehow the data is stil on tasmota. Also ive added 2/1/7 for the new action off but this doesn not get to this light but it does to others so the knx signal is there.

ascillato, 2021-01-29:
ok, please,

* go to the console and type ` weblog 4`
* then go to the KNX menu, and post here the screenshot of it.
* then on the KNX menu click on save
* Go again to the console and post the console output.

Thanks.

AhmedIbrahim88, 2021-10-31:
@ascillato, thank you a lot for all the help in this thread, I went through it but please allow me to ask a question differently just to be sure I got it the correct way .. I currently have KNX devices that are wired to the bus .. and I intend to install a relay that runs on tasmota along with a few push buttons so that I'll be doing is adding an IP interface to connect my existing KNX devices to IP after that configure tasmota now when I need another none tasmota device lets say a KNX wall switch to control the Tasmota relay do I add the tasmota as a dummy device on ETS ? and that would work ?

AhmedOsama, 2021-10-31:
@ascillato , thank you a lot for all the help in this thread, I went through it but please allow me to ask a question differently just to be sure I got it the correct way .. I currently have KNX devices that are wired to the bus .. and I intend to install a relay that runs on tasmota along with a few push buttons so that I'll be doing is adding an IP interface to connect my existing KNX devices to IP after that configure tasmota now when I need another none tasmota device lets say a KNX wall switch to control the Tasmota relay do I add the tasmota as a dummy device on ETS ? and that would work ?

MGKQB, 2021-10-31:
this issue is probably not the best place to discuss it but i'll answer anyway. (ascillato is busy enough)

first you need an KNX/IP router to connect your KNX-TP (twisted pair) network to IP. Make sure KNX routed traffic on IP Multicast 224.0.23.12 UDP port 3671 works between your tasmota devices and the KNX/IP router. That is the most problematic part for many poeple, some routers, switches and wireless access points do strange things with multicast or block it.

If IP works then you have pretty much two ways to make communication work for KNX. 

1) configure you KNX/IP router to pass ALL datagrams. This is kind of dangerous (IP could overwhelm the 9k6 TP network) but in a private home probably ok. NO dummy is required in ETS but it make your life easier to use any anyway.
2) your KNX/IP router filters and only necessary datagrams for forwarded between IP and TP. in this case you NEED the dummy because ETS "programs" the router filter tables.

AhmedOsama, 2021-10-31:
@MGKQB thank you

ascillato, 2021-10-31:
Hi,

Thanks for answering. That part of the setup is not fully explained in the Tasmota KNX docs ( https://tasmota.github.io/docs/KNX/ ). Please, if you could add more info there, that will help people a lot.

Thanks. very appreciated