ascillato/Tasmota_KNX #580: No KNX TX/RX/SCENE available when compiled with scripting enabled instead of rules

mwerski, 2021-04-11:
### PROBLEM DESCRIPTION
As mentioned in the subject. Probably caused by multiple "#ifdef USE_RULES" directives in xdrv_11_knx.ino. As scripting and rules exclude each other, it is not possible to use KNX commands from scripts (for example to send sensor values).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1
- [x] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Enable scripting instead of rules and build the firmware.

### EXPECTED BEHAVIOUR
KNX TX/RX/SCENE should also be available when using scripts, so that custom KNX commands can be executed.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-04-12:
Hi, yes, as you have discovered, there is no support for KNX if using Scripting.

ascillato, 2021-04-12:
Please, add your request to Ideas in Discussions of Tasmota ( https://github.com/arendst/Tasmota/discussions )

closed 2021-04-12 by ascillato

ascillato, 2021-04-13:
Please, add your request in the main Tasmota Repository, as in there it will have a wider audience than in my fork. Your request is good.

ascillato, 2021-04-14:
Thanks for posting this in Tasmota main repository. Very appreciated.