ascillato/Tasmota_KNX #597: Problem sending KNX telegrams

Sansibari, 2023-05-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Sansibari, 2023-05-05:
I configured a D1 8266 as following:

[image]

I use a MDT IP Gateway and ETS6
Have no problem to communicate via Node-Red over Wifi to the KNX Bus.
But no messages are received from the Tasmota Device.
The ETS Group monitor shows:

[image]
It seems, the DPT and Info is not received in the correct format.

Any ideas?
Thanks in advance!

Sansibari, 2023-05-05:
[image]

Sansibari, 2023-05-05:
[image]

Sansibari, 2023-05-05:
[image]

ascillato, 2023-11-16:
Hi, sorry to hear about your issue. Seems that your system is expecting the temperature in another format. Previously we were using that format but then we changed to a new one by users requests. Since we have received reports that some system uses the previous format we were thinking on adding an option to select the format in the UI. That requires some changes that at the moment I won't be able to work on. Hopefully in the future I have time to expand the KNX driver for Tasmota.

closed 2023-11-16 by ascillato